fastmachinelearning/hls4ml #226: hls4ml 0.3.0 with pip for Python 2.x gives error

Tommaso93, 2020-10-19:
Hello, I tried to install the hls4ml v0.3.0 using the `pip install hls4ml` command (with python 2.x).
During the installation I had the following error:
`
with open(path.join(this_directory, 'README.md'), encoding='utf-8') as f:
TypeError: 'encoding' is an invalid keyword argument for this function
`
In order to fix this, I manually downloaded the source code and modified the setup.py, changing 
`
open(path.join(this_directory, 'README.md'), encoding='utf-8')
`
with
`
import io
io.open(path.join(this_directory, 'README.md'), encoding='utf-8')
`
And then running the pip command inside the source code.

Tommaso93, 2020-10-19:
Edit: After this procedure, several import errors show up when executing:
```
import hls4ml
```

Duchstf, 2020-10-19:
Can you tried commenting those lines out and do `pip install .` from the source code main directory? Also what are the errors when importing?

vloncar, 2020-10-19:
Python 2 is not supported. We dropped compatibility before 0.2.0. Python 2 language is EOL since 1.1.2020., you should consider upgrading your flow to Python3.

Tommaso93, 2020-10-19:
Ok, no problem! I saw the compatibility in the readme, so I though that it could work! 

Thank you,
Tommaso

vloncar, 2020-10-19:
@Tommaso93 Where did you find the compatibility note in the documentation? We may have missed that one ;-)

Tommaso93, 2020-10-19:
In the python badge on the readme. Also in the pypi website.

vloncar, 2020-10-19:
Thanks, we'll update those!

Duchstf, 2020-10-19:
@vloncar I think it's because of these lines:

https://github.com/hls-fpga-machine-learning/hls4ml/blob/63b12c1493e118f92bdd4fff18aafdc99e1e0de0/setup.py#L39-L40

I can make a PR to fix this later.

closed 2021-09-13 by jmduarte