fastmachinelearning/hls4ml #227: hls4ml v0.3.0 with Python 3.x on MacOS 10.15.7 crashes

Tommaso93, 2020-10-19:
Hello,

I have installed hls4ml version 0.3.0 on Python 3.7.9 with MacOS 10.15.7 
Importing the package hls4ml, provide the following crash message:
```
[libprotobuf ERROR google/protobuf/descriptor_database.cc:394] Invalid file descriptor data passed to EncodedDescriptorDatabase::Add().
[libprotobuf FATAL google/protobuf/descriptor.cc:1356] CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size):
libc++abi.dylib: terminating with uncaught exception of type google::protobuf::FatalException: CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size):
Abort trap: 6
```
Looking on the internet, I found several topics suggesting to uninstall and reinstall tensorflow and protobuf but it does not work on my system. The older 0.2.0 version works fine.

I have pip 20.2.3 and:
- Tensorflow 2.3.1
- protobuf 3.13.0

Any suggestions?

Tommaso93, 2020-10-19:
Another user also has this issue:
https://github.com/hls-fpga-machine-learning/hls4ml/issues/217

vloncar, 2020-10-19:
Can you import tensorflow?

BTW, Vivado/Vitis is only available on Windows and Linux (of which we only truly support the latter), on Mac you can only use hls4ml to generate HLS projects, but you can't do anything with them.

Tommaso93, 2020-10-20:
Yes I can import Tensorflow. I know that Vivado is only available on Windows and Linux but, for the moment, I only need the HLS project that are then loaded in another linux machine.

vloncar, 2020-10-20:
We don't use `protobuf` directly in any code, so this can only come from importing either Tensorflow or ONNX. Both are optional. You can try setting up a conda/virtualenv without them and try that way. If you have a Keras model saved in h5 file, you don't need an installation of TF. Sorry, I can't reproduce this in Linux so it is not easy to debug.

jmduarte, 2021-09-12:
with hls4ml 0.5.0 and TensorFlow 2.6.0 on Mac OS X 11.5.2, this no longer seems to be an issue.

```
conda create -n hls4ml-test python=3.8
conda activate hls4ml-test
pip install tensorflow hls4ml
```

pip freeze:
```
absl-py==0.13.0
astunparse==1.6.3
cachetools==4.2.2
certifi==2021.5.30
charset-normalizer==2.0.4
clang==5.0
flatbuffers==1.12
gast==0.4.0
google-auth==1.35.0
google-auth-oauthlib==0.4.6
google-pasta==0.2.0
grpcio==1.40.0
h5py==3.1.0
hls4ml==0.5.0
idna==3.2
keras==2.6.0
Keras-Preprocessing==1.1.2
Markdown==3.3.4
numpy==1.19.5
oauthlib==3.1.1
onnx==1.10.1
opt-einsum==3.3.0
protobuf==3.17.3
pyasn1==0.4.8
pyasn1-modules==0.2.8
PyYAML==5.4.1
requests==2.26.0
requests-oauthlib==1.3.0
rsa==4.7.2
six==1.15.0
tensorboard==2.6.0
tensorboard-data-server==0.6.1
tensorboard-plugin-wit==1.8.0
tensorflow==2.6.0
tensorflow-estimator==2.6.0
termcolor==1.1.0
typing-extensions==3.7.4.3
urllib3==1.26.6
Werkzeug==2.0.1
wrapt==1.12.1
```

closed 2021-09-12 by jmduarte