fastmachinelearning/hls4ml #228: Inconsistent output of hls_model.predict()

Mmiglio, 2020-10-20:
Hi all, 

I'm trying to convert my model with HLS4ML but I'm facing some issues. 
A notebook reporting a minimal example can be found [here](https://gist.github.com/Mmiglio/846b793168e8ac9dff2993750468394d). 

The first problem I'm observing is that after converting the model from Keras, `hls_model.predict()` gives wrong predictions and different every time (this can be seen in the last two cells of the notebook I've attached). From my understanding the output should be deterministic, e.g. for a given input `x` there should be only one `f(x)`. Am I missing something? 

Furthermore, I'm able to build the model with `hls_model.build()`, but I'm not entirely convinced by the output report. 
[image]

The synthesis took around 45min, but me it looks like only the last layer has been correctly converted (dense + sigmoid activation). Am I doing something wrong? this could be why I'm seeing the wrong output. 

Thanks in advance for your help!

Mmiglio, 2020-10-21:
I managed to solve this problem by using the environment used in the tutorial: https://github.com/hls-fpga-machine-learning/hls4ml-tutorial/blob/fpl2020/environment.yml 

Now output and synthesis make sense:

[image]

Was this "silent" failure due to the wrong TensorFlow version? I was using TensorFlow 2.3.1

vloncar, 2020-10-21:
Hi @Mmiglio, glad to see that the issue is gone now, but the cause worries me. Can you send me the generated project from the previous case when it showed this strange behavior? (The contents of `model_cnn1D/hls4ml_prj` directory)

Mmiglio, 2020-10-21:
Hi @vloncar, you can find the content of that directory here (https://www.transfernow.net/NxPclG102020). It has a slightly different name because I performed several tests, but the content is the same (also the output). 
I'm interested in understanding the problem, let me know if you manage to find it or you need something else

husnainmubarik, 2020-10-22:
I think I'm facing the same issue - Keras and hls4ml accuracies do not match for me - 
[image]

And my vivado log file says - input1_V is dangling input 
myproject/input1_V' has no fanin or fanout and is left dangling

Duchstf, 2020-10-22:
Just to add to the turmoil :) 

My Conv1D model never reaches good accuracy as well (even if I use stable sigmoid activation). 

[image]

srizwantariq, 2020-10-23:
When i first get to know about HLS4ML, i was so happy to actually find out an automated way of implementing NN on FPGA. But currently i am also facing the problem of inconsistent results.

thesps, 2020-10-23:
Hi @husnainmubarik, I suspect this is something different, actually. Could you let me know which hls4ml and Tensorflow versions you're using? And if possible, could you send the `.tar.gz` archive that hls4ml creates of the HLS project?

 @Duchstf achieving good accuracy using fixed-point data types can be tricky. That's why we introduced the `profiling` and `trace` utilities that you can learn more about [here](https://fastmachinelearning.org/hls4ml/api/PROFILING.html]). You can also use things like regularization and BatchNormalization to bring your weights and activations into a more consistent range over the model. That said, this is not necessarily the same issue as what's seen here - getting different results each time from `predict`. Could you perhaps check whether you see that specific problem, or are your results consistent? If it's a different issue, you should isolate where the differences between Keras and hls4ml first occur - which layer - using `trace`, and try to tweak your data types. If you still see differences that can't be resolved, you should raise a new issue. 

@srizwantariq if you also see this issue of getting different `predict` results from the same input, could you possibly also provide an example?

Mmiglio, 2020-10-23:
Hi all, 
as discussed with Thea I'm providing the code to reproduce the performance problem. You can find everything reported in this notebook https://github.com/Mmiglio/HLS4ML_issue/blob/main/model_3classes.ipynb

husnainmubarik, 2020-10-23:
I'm using hls4ml 0.3.0 and tensorflow 2.3.1
My issue is three fold 
1) For the given 3 layers KERAS network hls4ml generated accuracy is low (0.20 for the hls4ml predict).
2) Input1_V port is not connected to anything in the RTL which means my input port is actually dandling input port log shown in the attached image. 
3) I've a question -> can I set unrolling in such a way that It only unrolls all the weight parameters and not the input parameters? In this way there'd be a unique DSP for every weight and not for every operation?
I've attached the link to the .tar.gz here  https://drive.google.com/drive/folders/1aeq6Wu7CfU91V1wSBiwFzJt4Vb7Q3p5k?usp=sharing
[image]
[image]
[image]

husnainmubarik, 2020-10-23:
I used the environment from here [ENV](https://github.com/hls-fpga-machine-learning/hls4ml-tutorial/blob/fpl2020/environment.yml) and my hls4ml accuracy matched with the keras accuracy.

thesps, 2020-10-24:
I think @Mmiglio 's problem is the same TF 2.3 input layer issue we saw before.
I see in the project archive you sent that the model has function signature:
```
void project_cnn_1d(
    input_t input1[N_INPUT_1_1*N_INPUT_2_1], // this one in particular
    result_t layer9_out[N_LAYER_8],
    unsigned short &const_size_in_1,
    unsigned short &const_size_out_1
) {
```
but then shortly afterwards defines
```
    input2_t input_1[N_INPUT_1_2*N_INPUT_2_2];
    // lines removed for clarity
    nnet::normalize<input2_t, layer3_t, config3>(input_1, layer3_out, s3, b3); // the first layer of the model
```
So, the actual model input array is never read, and you see different results each time because the array `input_1` is fed into the first network layer without initialization, so you just get whatever garbage was already there. 
(Note `input1` is not the same as `input_1`).

I think this is actually the same issue as @husnainmubarik was seeing, and was fixed [here](https://github.com/hls-fpga-machine-learning/hls4ml/commit/9305d0ccef8b85f55aae3639b149d62c3f49d020). So, the solution is to use the master branch, and for us to make a new release soon!

thesps, 2020-11-02:
We released hls4ml v0.4.0 last Friday, which should resolve the issues seen here. Please try it out and let us know.

Mmiglio, 2020-11-06:
I tested it with tensorflow `2.3.1` and now it works! thanks for this new release.

closed 2020-11-06 by Mmiglio