fastmachinelearning/hls4ml #233: failed to compile project

mechlya, 2020-10-29:
Hello
I tryed  to compile the project but this process gave me that error.

print(config)
Interpreting Sequential
Topology:
Layer name: input_1, layer type: InputLayer
Layer name: conv2d, layer type: Conv2D
  -> Activation (relu), layer name: conv2d
Layer name: conv2d_1, layer type: Conv2D
  -> Activation (relu), layer name: conv2d_1
Layer name: max_pooling2d, layer type: MaxPooling2D
Layer name: conv2d_2, layer type: Conv2D
  -> Activation (relu), layer name: conv2d_2
Layer name: max_pooling2d_1, layer type: MaxPooling2D
Layer name: conv2d_3, layer type: Conv2D
  -> Activation (relu), layer name: conv2d_3
Layer name: max_pooling2d_2, layer type: MaxPooling2D
Layer name: conv2d_4, layer type: Conv2D
  -> Activation (relu), layer name: conv2d_4
Layer name: max_pooling2d_3, layer type: MaxPooling2D
Layer name: conv2d_5, layer type: Conv2D
  -> Activation (relu), layer name: conv2d_5
Layer name: max_pooling2d_4, layer type: MaxPooling2D
Layer name: conv2d_6, layer type: Conv2D
  -> Activation (relu), layer name: conv2d_6
Layer name: max_pooling2d_5, layer type: MaxPooling2D
Layer name: dense, layer type: Dense
  -> Activation (relu), layer name: dense
Layer name: dense_1, layer type: Dense
  -> Activation (sigmoid), layer name: dense_1
{'Model': {'Precision': 'ap_fixed<16,6>', 'ReuseFactor': 1, 'Strategy': 'Latency'}}

hls_model = hls4ml.converters.convert_from_keras_model(model, hls_config=config, output_dir='D:/Vivado_HLS_CNN/HLS_CNN/test2')
Interpreting Sequential
Input shape: [256, 256, 3]
Topology:
Layer name: input_1, layer type: InputLayer, current shape: [[None, 256, 256, 3]]
Layer name: conv2d, layer type: Conv2D, current shape: [[None, 256, 256, 3]]
Layer name: conv2d_1, layer type: Conv2D, current shape: [[None, 256, 256, 32]]
Layer name: max_pooling2d, layer type: MaxPooling2D, current shape: [[None, 256, 256, 64]]
Layer name: conv2d_2, layer type: Conv2D, current shape: [[None, 128, 128, 64]]
Layer name: max_pooling2d_1, layer type: MaxPooling2D, current shape: [[None, 128, 128, 128]]
Layer name: conv2d_3, layer type: Conv2D, current shape: [[None, 64, 64, 128]]
Layer name: max_pooling2d_2, layer type: MaxPooling2D, current shape: [[None, 64, 64, 160]]
Layer name: conv2d_4, layer type: Conv2D, current shape: [[None, 32, 32, 160]]
Layer name: max_pooling2d_3, layer type: MaxPooling2D, current shape: [[None, 32, 32, 192]]
Layer name: conv2d_5, layer type: Conv2D, current shape: [[None, 16, 16, 192]]
Layer name: max_pooling2d_4, layer type: MaxPooling2D, current shape: [[None, 16, 16, 256]]
Layer name: conv2d_6, layer type: Conv2D, current shape: [[None, 8, 8, 256]]
Layer name: max_pooling2d_5, layer type: MaxPooling2D, current shape: [[None, 8, 8, 320]]
Layer name: dense, layer type: Dense, current shape: [[None, 4, 4, 320]]
Layer name: dense_1, layer type: Dense, current shape: [[None, 128]]
Creating HLS model

hls_model.compile()
Writing HLS project
Done
Traceback (most recent call last):

  File "<ipython-input-10-20374fcfe547>", line 1, in <module>
    hls_model.compile()

\hls_model.py", line 369, in compile
    raise Exception('Failed to compile project "{}"'.format(self.config.get_project_name()))

Exception: Failed to compile project "myproject"

vloncar, 2020-10-29:
Windows is not supported.

mechlya, 2020-10-29:
it generated  those files

firmware    
tb_data
build_lib.sh
build_prj.tcl
myproject_bridge.cpp
myproject_test.cpp
vivado_synth.tcl

can I use it to synthesise the project on vivado hls and creat an ip !

vloncar, 2020-10-29:
yes, but not on Windows

closed 2020-12-07 by thesps