fastmachinelearning/hls4ml #234: LeNet gives poor accuracy.

husnainmubarik, 2020-11-01:
I tried LeNet with mnist dataset; and the accuracy of hls4ml model is 0.25 whereas accuracy of keras model is 0.96. 
Any idea how can I fix that?
[image]
[image]

thesps, 2020-11-02:
This could happen due to choice of fixed-point data types not working well for your model. Did you try any of the `profiling` or `trace` functionality, or tune data types according to your model? 

Since you mentioned MNIST, I think it's relatively likely that you'll need to use the 'Stable' implementation of the Softmax activation (see #207) for details. You can do this by setting something like `config['LayerName']['softmax']['Strategy'] = 'Stable'` in your hls4ml configuration, though I'm not sure what your final activation layer name is, so replace `'softmax'` with whatever that is.

Mmiglio, 2020-11-02:
Hi @thesps, I'm facing a similar problem. Looking at `profiling` functionality output I see that I may have problems with the `softmax` layer. 

[image]
Even after tuning the data types (I'm using `ap_fixed<50,5>` to "cover" the `softmax` activation) I can't get even close to my model performance (0.60 vs 0.97 accuracy). Is there something I can do at training level / after training to mitigate this problem ?

thesps, 2020-11-16:
@Mmiglio did you also try the 'Stable' softmax (like `config['LayerName']['softmax']['Strategy'] = 'Stable'`) ?

closed 2026-10-27 by github-actions[bot]