fastmachinelearning/hls4ml #235: hls4ml --version gives 0.2.0 even if it is 0.4.0

SerenaC94, 2020-11-03:
Hi,
this is just a microscopic issue, but since I spent some time figuring out what was wrong, maybe it can be useful to others as well.
I am following the tutorials in [https://github.com/hls-fpga-machine-learning/hls4ml-tutorial](https://github.com/hls-fpga-machine-learning/hls4ml-tutorial), I used conda to set up the environment.

`environment.yml` states that hls4ml version 0.4.0 is needed, however running
`hls4ml --version`
the answer is
`hls4ml 0.2.0`

I thought there were problems with conda, but then finally I saw that it is just that [here](https://github.com/hls-fpga-machine-learning/hls4ml/blob/v0.4.0/scripts/hls4ml) the version is stuck at 0.2.0!

thesps, 2020-11-03:
Good spot, thanks!

jmduarte, 2020-11-03:
@thesps we could use something like this https://github.com/peritus/bumpversion

closed 2021-07-16 by jmduarte