fastmachinelearning/hls4ml #236: Pre-synthesis fails on hls4ml 0.4.0

rajeshpandit107, 2020-11-05:
[image]

Synthesizing a dense model for mnist with 128*128*128*10 architecture with a reuse factor of 200. Pre-synthesis fails

nhanvtran, 2020-11-05:
What did you set as the `Strategy`?  For a model of this size you should use `Resource`

rajeshpandit107, 2020-11-06:
I used the strategy as resource, getting the same error still.
[image]

vloncar, 2020-11-06:
CNNs in 0.4.0 work only for very small models. You'll have to wait for 0.5.0 which will come after #220 is merged

rajeshpandit107, 2020-11-06:
synthesis completes with version 0.3.0 with same configuration, could you please suggest the solution?

vloncar, 2020-11-06:
How do you set the Strategy to "resource"? I can't suggest a solution if you don't provide the code that fails.

rajeshpandit107, 2020-11-06:
config['Model']['Strategy'] = 'resource'

vloncar, 2020-11-06:
It works for me.

>  I can't suggest a solution if you don't provide the code that fails.

^ this

rajeshpandit107, 2020-11-11:
Hi, 
I am providing the jupter-notebook code and vivado_hls log file. I have an architecture of 128*128*128*10 . Pre-synthesis fails on loop unrolling. Could you please provide the solution?

Thanks.
[vivado_hls.log](https://github.com/fastmachinelearning/hls4ml/files/5522459/vivado_hls.log)
[jupyter_notebook_code.txt](https://github.com/fastmachinelearning/hls4ml/files/5522460/jupyter_notebook_code.txt)

rajeshpandit107, 2020-11-11:
I increased the reuse factor to 2000 and it worked.

closed 2020-11-11 by rajeshpandit107