fastmachinelearning/hls4ml #237: Generate test-bench data

jpazzini, 2020-11-06:
Hi,

after assessing the performances of our model with `hls_model.predict()` we'd now like to generate testbench data files.

As far as we've seen from the code this should be implemented in the `write_test_bench` [function](https://github.com/hls-fpga-machine-learning/hls4ml/blob/master/hls4ml/writer/vivado_writer.py#L320), but neither [InputData](https://github.com/hls-fpga-machine-learning/hls4ml/blob/master/hls4ml/writer/vivado_writer.py#L330) nor [OutputPredictions](https://github.com/hls-fpga-machine-learning/hls4ml/blob/master/hls4ml/writer/vivado_writer.py#L331) seems to point to a valid key of the `model.config`.

Indeed, nothing gets generated as testbench data in the `tb_data` folder once the project is compiled.

Is there a way to generate tb_data automatically from hls4ml?

Thanks

vloncar, 2020-11-06:
You need to create those files yourself and add the keys to the `model.config.config` dict if you want them in the test bench. The format is simple, file should have `.dat` extension, contain one sample per line, and features are space-separated.

For example:
```
import numpy as np

arr = np.zeros((12,34,56))
arr = arr.reshape(-1, np.prod(arr.shape[1:]))

with open('my_file.dat', 'w') as my_file:
    for sample in arr:
        my_file.write(' '.join(str(x) for x in sample))
        my_file.write('\n')
```

closed 2020-11-23 by jpazzini

Duchstf, 2020-11-23:
btw, a simple way to generate the test bench files is to just save the array as a `.npy` file. So:

```
import numpy as np

arr = np.zeros((12,34,56))

np.save("test_arr.npy", arr)
```

And then pass the `npy` files into the InputData/OutputPredictions config.