fastmachinelearning/hls4ml #238: h5py version issue causes model conversion to fail

veyron8800, 2020-11-09:
Hello,

When installing a fresh copy of hls4ml, I receive the following error when attempting to convert a model. (Please ignore the version name in the directory, I installed the latest version through pip)
```
 /home/areis/Desktop/HLS4MLV3/venv/bin/hls4ml:166: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if found is not 0:
Loading configuration from ExModel.yml
Traceback (most recent call last):
  File "/home/areis/Desktop/HLS4MLV3/venv/bin/hls4ml", line 187, in <module>
    main()
  File "/home/areis/Desktop/HLS4MLV3/venv/bin/hls4ml", line 67, in main
    args.func(args)
  File "/home/areis/Desktop/HLS4MLV3/venv/bin/hls4ml", line 136, in convert
    model = hls4ml.converters.convert_from_yaml_config(yamlConfig)
  File "/home/areis/Desktop/HLS4MLV3/venv/lib/python3.8/site-packages/hls4ml/converters/__init__.py", line 61, in convert_from_yaml_config
    model = keras_to_hls(yamlConfig)
  File "/home/areis/Desktop/HLS4MLV3/venv/lib/python3.8/site-packages/hls4ml/converters/keras_to_hls.py", line 151, in keras_to_hls
    model_arch = json.loads(model_arch.decode('utf-8'))
AttributeError: 'str' object has no attribute 'decode'
```

This error is fixed by downgrading h5py to 2.10.0

Thanks

Duchstf, 2020-11-09:
@veyron8800 Can you send me the model?

veyron8800, 2020-11-10:
@Duchstf I can consistently recreate this bug using any model when I save the entire model to a h5 file using tf keras' `model.save()`.  The specific files that relate back to the error output above are the same ones included in #239 .

Duchstf, 2020-11-10:
Ah I see, `tf.keras` just changed their saving mechanisms.

Duchstf, 2020-11-10:
Ok, for anyone encountering this issue, here is a quick workaround:

`pip install hls4ml h5py<3.0.0`

The new releases of `h5py` introduces this error, and it also breaks things in Tensorflow (tensorflow/tensorflow#44467). I'm waiting to see if there will be fixes from tensorflow or h5py side and adapt to it. Otherwise I'll just ping the h5py version to `h5py==2.10.0` in the next hls4ml release.

closed 2020-12-24 by Duchstf