fastmachinelearning/hls4ml #240: Feature Request: Depthwise Separable Convolution

veyron8800, 2020-11-09:
Hello,

In [this deck](https://lss.fnal.gov/archive/2019/slides/fermilab-slides-19-706-scd-v.pdf), depthwise separable convolution is mentioned as a feature that was being worked on. Was that work ever finished, or was it put aside for other additions?

Thanks

vloncar, 2020-11-09:
`DepthwiseConv2D` was recently added to the cnn_stream branch. See #220

veyron8800, 2020-11-10:
@vloncar , when attempting model conversion the cnn_stream branch, the function `get_valid_reuse_factors` is throwing an exception. `n_in` and `n_out` are only set for Dense, Conv1D, and Conv2D layers, so SeparableConv2D/DepthwiseConv2D layers cause an IndexError to be thrown on [line 401](https://github.com/vloncar/hls4ml/blob/11ac85bfeb674e6e4208bfe4516cfb76edcd3ffb/hls4ml/templates/vivado_template.py#L401).  Are reuse factors still valid in the new stream layer implementation paradigm?

vloncar, 2020-11-10:
`Resource` strategy doesn't exist (yet) for Deparable/Depthwise layers, you need to use `Latency`. You can override it in config. But yes, layers should be smarter about this, I'll fix this soon.

veyron8800, 2020-11-10:
@vloncar Sounds good. Another quick question, is the softmax activation supported with streaming? I'm getting the following error on csim:
```
cannot convert ‘layer7_out’ (type ‘hls::stream<nnet::array<ap_fixed<16, 6>, 1u> >’) to type ‘nnet::array<ap_fixed<16, 6>, 1u>*’
     nnet::sigmoid<layer7_t, result_t, sigmoid_config8>(layer7_out, layer8_out);
```

vloncar, 2020-11-10:
It's not supported right now, but it is really easy to implement. I was keeping the extra activations as a "good first issue" for newcomers :-)

veyron8800, 2020-11-16:
Makes sense. Thanks for your help!

closed 2020-11-16 by veyron8800