fastmachinelearning/hls4ml #242: Small Conv2D model synthesis running for days

bengook, 2020-11-11:
Hello,

I am attempting to synthesize my own small Conv2D keras model,
```
Model: "sequential"
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
conv2d (Conv2D)              (None, 508, 1, 16)        304       
_________________________________________________________________
activation (Activation)      (None, 508, 1, 16)        0         
_________________________________________________________________
dropout (Dropout)            (None, 508, 1, 16)        0         
_________________________________________________________________
conv2d_1 (Conv2D)            (None, 250, 1, 16)        2320      
_________________________________________________________________
activation_1 (Activation)    (None, 250, 1, 16)        0         
_________________________________________________________________
dropout_1 (Dropout)          (None, 250, 1, 16)        0         
_________________________________________________________________
conv2d_2 (Conv2D)            (None, 121, 1, 8)         1160      
=================================================================
Total params: 3,784
Trainable params: 3,784
Non-trainable params: 0
_________________________________________________________________
```
using hls4ml v0.4.0 and Vivado HLS 2017.2  (although I'm also trying 2018.2 and 2019.1 with the maximum partitions increased) on Ubuntu 18.04.5, and the vivado_hls C/RTL Synthesis has been stuck at the hardware synthesis for several days now. I've attached the [vivado_hls.log](https://github.com/fastmachinelearning/hls4ml/files/5525876/vivado_hls.log) file, and the keras-config.yml is:
```
KerasJson: sm_fullyconv2d_minimodrec.json
KerasH5:   sm_fullyconv2d_minimodrec.h5
#InputData: keras/KERAS_3layer_input_features.dat
#OutputPredictions: keras/KERAS_3layer_predictions.dat
OutputDir: hls-fullyCNN-minimod
ProjectName: myproject
XilinxPart: xc7k410tffg900-2 
#xcku115-flvb2104-2-i
ClockPeriod: 5

IOType: io_parallel # options: io_serial/io_parallel
HLSConfig:
  Model:
    Precision: ap_fixed<13,6>
    ReuseFactor: 64
    Strategy: Resource
```

I have successfully synthesized the example model, Keras_conv2d_model, using hls4ml v0.4.0 and Vivado HLS 2017.2 in a matter of minutes, which has on the order of the same number or parameters as my model, but I am out of ideas as to why my model gets stuck. Any help would be greatly appreciated. 

Thanks.

veyron8800, 2020-11-20:
Hello @bengook ,

This is an issue I have run into many times when trying to synthesize models using the master branch. The solutions which have worked for me are incredibly hacky, as they involve adjusting many pragmas and some of the code structure in the generated firmware. However, there is some great work being done on streaming implementations of the convolutional layers, which look incredibly promising. I would first give #220 a try to see if it would work with your model.

vloncar, 2020-11-20:
Thanks @veyron8800 for pointing this out, I forgot to answer this.

@bengook You can try the branch from #220. To install it, run `pip install git+https://github.com/vloncar/hls4ml.git@cnn_stream`. In your yml config, set `IOType: io_stream` and it should work.

bengook, 2020-11-23:
Thank you for the suggestion to us the cnn_stream branch. I've attempted it on my model, but now I am running into compiler errors in the csim_design step. It looks like an issue with the pixels:
```
../../../../firmware/parameters.h:159:0: error: conversion from '__int128' to 'ap_uint<9>' is ambiguous
const ap_uint<config8::filt_height * config8::filt_width> config8::pixels[] = {A LONG LIST OF VERY LARGE NUMBERS}
```
Is this something to do with the PackingFactor? And if so, do I set this similarly to the ReuseFactor in the yml config?
Are the limitations mentioned in #220 still affecting this branch, specifically the requirement for square kernels? Which Vivado should be used with this, 2017.2, 2018.1...?

Thank you again.

vloncar, 2020-11-23:
I haven't encountered that error yet. Is this still on linux?

As for the limitations, the square kernel one still exists, the others are mostly gone. I should really update the text of that PR. The 2017.2 is known not to work and I also remember some issues in 2018.2 but I don't know if they are relevant to the current code base. I tested mostly on 2020.1 and 2019.1.

bengook, 2020-11-23:
Yes, this is on Ubuntu 18.04. My model has rectangular kernels, I will modify it to have square ones and see if that changes anything, and I'll also use 2019.1.

bengook, 2020-11-23:
Fantastic! Once I changed my kernels to be square, #220 with Vivado 2019.1 works on my model! Now I just need to optimize my model such that it doesn't use 350% of the LUTs on my specific FPGA 😒 . Thank you very much for your help!

closed 2020-11-23 by bengook