fastmachinelearning/hls4ml #244: KeyError when compiling model with multiple inputs

DelonShen, 2020-11-17:
When trying to compile the following model with multiple inputs:
```
import numpy as np
from keras.models import Input, Model, load_model
import keras
import tensorflow as tf
import hls4ml

def build_model():
    input_mu = Input(shape =(1,))
    input_adc = Input(shape = (5,))
    sub_model_1 = keras.layers.Dense(16,  name="sub_model_1")(input_mu)
    sub_model_2 = keras.layers.Dense(16,  name="sub_model_2")(sub_model_1)
    sub_model_3 = keras.layers.Dense(5,  name="sub_model_3")(sub_model_2)
    output = keras.layers.Add()([sub_model_3, input_adc]) 
    return Model(inputs = [input_mu, input_adc], outputs = output)

model = build_model()

config = hls4ml.utils.config_from_keras_model(model, granularity='model')

hls_model = hls4ml.converters.convert_from_keras_model(model, hls_config = config, output_dir='test_hls')
hls_model.compile()
```

I get the following output:
```
Interpreting Model
Topology:
Layer name: input_1, layer type: InputLayer
Layer name: sub_model_1, layer type: Dense
  -> Activation (linear), layer name: sub_model_1
Layer name: sub_model_2, layer type: Dense
  -> Activation (linear), layer name: sub_model_2
Layer name: sub_model_3, layer type: Dense
  -> Activation (linear), layer name: sub_model_3
Layer name: input_2, layer type: InputLayer
Layer name: add, layer type: Add
Interpreting Model
Topology:
Layer name: input_1, layer type: InputLayer, current shape: [[None, 1]]
Layer name: sub_model_1, layer type: Dense, current shape: [[None, 1]]
Layer name: sub_model_2, layer type: Dense, current shape: [[None, 16]]
Layer name: sub_model_3, layer type: Dense, current shape: [[None, 16]]
Layer name: input_2, layer type: InputLayer, current shape: [[None, 5]]
Layer name: add, layer type: Merge, current shape: [[None, 5]]
Creating HLS model
Writing HLS project
Traceback (most recent call last):
  File "hls4ml_scratchbook.py", line 23, in <module>
    hls_model.compile()
  File "/home/delon/.anaconda3/envs/research/lib/python3.8/site-packages/hls4ml/model/hls_model.py", line 361, in compile
    self.write()
  File "/home/delon/.anaconda3/envs/research/lib/python3.8/site-packages/hls4ml/model/hls_model.py", line 358, in write
    self.config.writer.write_hls(self)
  File "/home/delon/.anaconda3/envs/research/lib/python3.8/site-packages/hls4ml/writer/vivado_writer.py", line 582, in write_hls
    self.write_project_cpp(model)
  File "/home/delon/.anaconda3/envs/research/lib/python3.8/site-packages/hls4ml/writer/vivado_writer.py", line 173, in write_project_cpp
    func = layer.function_cpp()
  File "/home/delon/.anaconda3/envs/research/lib/python3.8/site-packages/hls4ml/model/hls_layers.py", line 752, in function_cpp
    params['input1_t'] = self.get_input_variable(self.inputs[0]).type.name
  File "/home/delon/.anaconda3/envs/research/lib/python3.8/site-packages/hls4ml/model/hls_layers.py", line 256, in get_input_variable
    return self.model.get_layer_output_variable(input_name)
  File "/home/delon/.anaconda3/envs/research/lib/python3.8/site-packages/hls4ml/model/hls_model.py", line 355, in get_layer_output_variable
    return self.output_vars[output_name]
KeyError: 'sub_model_3_linear'
```

Is there a way to deal with this?

vloncar, 2020-11-17:
Can you try using #220 branch? Or add a non-linear activation to those dense layers. This issue could be triggered by the removal of linear activation from the model graph and master branch doesn't handle this well with multiple inputs. This was recently fixed in #220.

DelonShen, 2020-11-19:
Hi, adding non-linear activation fixed the issue.

closed 2020-11-19 by DelonShen