fastmachinelearning/hls4ml #251: IP Core control interface

KOVI89alipes, 2020-11-30:
First, I would like to thank to HLS4ML team for their contribution in the world of FPGA based machine learning world. Your project is really fascinating

Initially, I tried to integrate HLS4ML products (HLS projects) into Vitis Accell flow and it works pretty well, but due to some technical issues, we had to make a step back to classic, Vivado design flow, in a block design. 
I added the respective directives, switched both input and output ports to AXI-LIte so now it's possible to load and read the data as axi memory mapped cells and the IP core was integrated into existing block design with PCIe XDMA.

But the control interface, group purple lines on the waveform below, is still floating. 
[image]
In a block design I temporarily connected them to the AXI GPIO kernel, but it's definitely ugly :)

As soon as I'm going to work with this kernel in a stream mode, I can imagine that I should write some very basic RTL kernel, which will `ap_start` strobe every time, when the last register of input is written. 

But if I really have to reinvent the wheel? Probably you already have some way of how to integrate HLS4ML products into Vivado Block design?

Thank you,

KOVI89alipes, 2020-11-30:
Dirty solution with GPIO

[image]

thesps, 2020-11-30:
Hi, thanks for the kind words. Can I ask which device you're targeting? We have a few developments for connecting hls4ml IPs into specific types of devices, for example for Alveo or Zynq SoC.

KOVI89alipes, 2020-12-01:
Hi,
We are targeting Alveo U50 card and more specific `xilinx_u50_qsfp_v1_1 ` block-design from Xilinx.
This design includes XDMA IP Core, and so we have `AXI` and `AXI_LITE` interface available.

thesps, 2020-12-07:
I'm closing this issue since the conversation has moved to #254

closed 2020-12-07 by thesps