fastmachinelearning/hls4ml #252: support for vivado 2020 or vitis 2020

simplelins, 2020-12-03:
hi, I try to use hls4ms with the model "KERAS_conv1d.json" at vitis 2020 ,and I want to know whether all the models at folder "hls4ml/example-models/keras" that have been tested and correctly. for me I am working for supporting vitis 2020 and got something error but no error log showing,  so i test the model "KERAS_conv1d.json" at vivado 2020 firstly and also failed.
 I suspect that the error caused by the different version between vivado 2019 and vivado 2020. someone can help me , thanks

**for vivado 2020, the error as below:**

> INFO: [HLS 200-489] Unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense.h:160) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config8>' completely with a factor of 3200.
ERROR: [XFORM 203-504] Stop unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense.h:160) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config8>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

**for vitis 2020, the error as below:**
***** C/RTL SYNTHESIS *****
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ... 
WARNING: [HLS 214-104] Only for-loops and functions support the dataflow: firmware/nnet_utils/nnet_dense.h:151:9
Resolution: For help on HLS 214-104 see www.xilinx.com/html_docs/xilinx2020_1/hls-guidance/214-104.html
WARNING: [HLS 214-104] Only for-loops and functions support the dataflow: firmware/nnet_utils/nnet_dense.h:151:9
Resolution: For help on HLS 214-104 see www.xilinx.com/html_docs/xilinx2020_1/hls-guidance/214-104.html
WARNING: [HLS 200-471] Dataflow form checks found 2 issue(s) in file firmware/myproject.cpp
Resolution: For help on HLS 200-471 see www.xilinx.com/html_docs/xilinx2020_1/hls-guidance/200-471.html
Compilation of the preprocessed source 'myproject' failed
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vitis_hls at Thu Dec  3 13:04:23 2020...

and python script as below:
`config = hls4ml.utils.fetch_example_model('KERAS_conv1d.json',"vitis")
print("-----------------config-----------------------")
print(config) #You can print it to see some default parameters
print("-----------------config finish-----------------------")
#Convert it to a hls project
hls_model = hls4ml.converters.keras_to_hls(config)

# Print full list of example model if you want to explore more
# hls4ml.utils.fetch_example_list()
print(hls_model.config.get_output_dir())
hls_model.write()
hls_model.build(csim=False, export=True)
`

jmduarte, 2021-03-03:
Started a branch to try to make some trivial changes to use the `vitis_hls` executable (https://github.com/fastmachinelearning/hls4ml/compare/master...jmduarte:vitis), but quickly found there are many issues to be fixed with pragmas that changed, etc.

e.g. running the baseline [hls4ml-tutorial](https://github.com/fastmachinelearning/hls4ml-tutorial) model, I get errors below:
```
****** Vitis HLS - High-Level Synthesis from C, C++ and OpenCL v2020.2 (64-bit)
  **** SW Build 3064766 on Wed Nov 18 09:12:47 MST 2020
  **** IP Build 3064653 on Wed Nov 18 14:17:31 MST 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

source /xilinx/Vitis_HLS/2020.2/scripts/vitis_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/xilinx/Vitis_HLS/2020.2/bin/unwrapped/lnx64.o/vitis_hls'
INFO: [HLS 200-10] For user 'root' on host 'prp-gpu-1.t2.ucsd.edu' (Linux_x86_64 version 3.10.0-1127.19.1.el7.x86_64) on Wed Mar 03 10:33:18 PST 2021
INFO: [HLS 200-10] On os "CentOS Linux release 7.8.2003 (Core)"
INFO: [HLS 200-10] In directory 'hls4ml_prj'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-1510] Running: open_project myproject_prj 
INFO: [HLS 200-10] Opening project 'hls4ml_prj/myproject_prj'.
INFO: [HLS 200-1510] Running: set_top myproject 
INFO: [HLS 200-1510] Running: add_files firmware/myproject.cpp -cflags -std=c++0x 
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb myproject_test.cpp -cflags -std=c++0x 
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb firmware/weights 
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-1510] Running: add_files -tb tb_data 
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-1510] Running: open_solution solution1 
INFO: [HLS 200-10] Opening solution 'hls4ml_prj/myproject_prj/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [HLS 200-10] Setting target device to 'xcu250-figd2104-2L-e'
INFO: [HLS 200-1505] Using flow_target 'vivado'
Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2020.2;t=hls+guidance;d=200-1505.html
INFO: [HLS 200-1464] Running solution command: config_compile -name_max_length=60
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [HLS 200-1464] Running solution command: config_compile -complex-mul-dsp=0
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [HLS 200-1510] Running: config_array_partition -maximum_size 4096 
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
ERROR: [HLS 200-642] The 'config_array_partition -maximum_size' command is not supported.
INFO: [HLS 200-1510] Running: config_compile -name_max_length 60 
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [HLS 200-1510] Running: set_part xcu250-figd2104-2L-e 
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [HLS 200-1510] Running: create_clock -period 5 -name default 
***** C/RTL SYNTHESIS *****
INFO: [HLS 200-1510] Running: csynth_design 
INFO: [HLS 200-111] Finished File checks and directory preparation: CPU user time: 0.01 seconds. CPU system time: 0 seconds. Elapsed time: 0 seconds; current allocated memory: 191.185 MB.
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ... 
ERROR: [HLS 214-125] no member named 'vector' in namespace 'std': hls4ml_prj/firmware/nnet_utils/nnet_helpers.h:279
ERROR: [HLS 214-124] use of undeclared identifier 'src': hls4ml_prj/firmware/nnet_utils/nnet_helpers.h:279
ERROR: [HLS 214-123] expected unqualified-id: hls4ml_prj/firmware/nnet_utils/nnet_helpers.h:279
ERROR: [HLS 214-125] no member named 'vector' in namespace 'std': hls4ml_prj/firmware/nnet_utils/nnet_helpers.h:286
ERROR: [HLS 214-124] use of undeclared identifier 'src': hls4ml_prj/firmware/nnet_utils/nnet_helpers.h:286
ERROR: [HLS 214-124] use of undeclared identifier 'dst': hls4ml_prj/firmware/nnet_utils/nnet_helpers.h:286
ERROR: [HLS 214-123] expected unqualified-id: hls4ml_prj/firmware/nnet_utils/nnet_helpers.h:286
INFO: [HLS 200-111] Finished Command csynth_design CPU user time: 8.56 seconds. CPU system time: 0.87 seconds. Elapsed time: 9.09 seconds; current allocated memory: 192.769 MB.
Syntax check failed in clang-tidy
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [HLS 200-112] Total CPU user time: 11.85 seconds. Total CPU system time: 2.07 seconds. Total elapsed time: 11.81 seconds; peak allocated memory: 192.402 MB.
INFO: [Common 17-206] Exiting vitis_hls at Wed Mar  3 10:33:30 2021...
```

jmitrevs, 2021-03-03:
I think lots of these types of errors are due to not including proper header files. They maybe worked before because the header files were included by other includes, but not as versions changed.

jmduarte, 2021-05-25:
Adding here the main pragma changes with Vitis HLS 2020.2 (thanks to @vloncar) and implications for hls4ml.
Help and/or ideas are welcome for those interested in contributing to fixing some of these issues toward creating `VitisBackend`
​
Changes:
- `pipeline` pragma can only be applied to loops
  - Affects activations, all `Latency` strategy algorithms and bits of `io_stream` algorithms
- `array_partition`/`array_reshape` cannot be applied to top function arguments
  - This trips up the whole synthesis
  - Interface is defined differently now
- `allocation function` pragma doesn't accept templates anymore
  - Reuse factor doesn't work without this

Omissions:
- `data_pack` is gone now
  - Affects `io_stream` implementation
  - Potentially replaced with the new `aggregate` pragma
​
Deprecations:
- `inline region`
  - Without this, we will not be able to use existing `dense` functions in other layers (e.g., `Conv2D`)
  - No clear replacement, however `inline recursive` may be enough in this case
- `function_instantiate`
  - No direct replacement, for functions without arguments replaceable with `allocation` pragma 
- `resource`
  - Affects "store weights in BRAM" functionality
  - Could be replaced with the new `bind_storage` pragma

Bottom line:
- Major non-backward-compatible changes
- Supporting both `vivado_hls` and `vitis_hls` will be tricky
- We need to abstract the pragmas away
  - Similar to HLSLib from ETHZ folks?
  - Alternative is using the `directives.tcl` file, but this is not the preferred solution
​

First attempt at synthesis in Vitis HLS (making changes above):
- Even with all changes made by hand, QoR suffer:
  - The layers often fail to meet scheduling and/or have increased II
  - The final design has 10-20x larger latency
  - Some models don’t even get that far and get stuck in pre-synthesis step 
  - Only `Dense` layers with `RF=1` "work" (reach the finish line, with suboptimal results)

jeinstei, 2021-11-24:
I'm in to help with this. I'd really rather not have two tools installed unless it really becomes needed.

jmduarte, 2023-04-09:
Preliminary support now included by #629

closed 2023-04-09 by jmduarte