fastmachinelearning/hls4ml #261: Bug(?): Problem with config ['IOType'] = 'io_serial'

KOVI89alipes, 2020-12-11:
Code to reproduce the error with hls4ml v0.4
```python
import hls4ml
from keras.models import model_from_json
json_fname = 'KERAS_3layer.json'
config = hls4ml.utils.fetch_example_model(json_fname)
config ['IOType'] = 'io_serial'
hls_model = hls4ml.converters.keras_to_hls(config)
hls_model.write()
```
Error:
```
  File "~/.local/lib/python3.8/site-packages/hls4ml/writer/vivado_writer.py", line 93, in _make_array_pragma
    return '#pragma HLS STREAM variable={name} depth={depth} dim={dim}'.format(name=variable.name, depth=depth, dim=0)
UnboundLocalError: local variable 'depth' referenced before assignment
```
https://github.com/fastmachinelearning/hls4ml/blob/521deb1a9929588fd57e1ca5e4255fbb0cff1b8f/hls4ml/writer/vivado_writer.py#L93

What should be the proper way to enable `saxi` interface for the input and output layers in the model configuration?

Now I'm just replacing in `model_name.cpp`
```
<<<    #pragma HLS INTERFACE ap_vld port=input_1,layer9_out 
>>>    #pragma HLS INTERFACE axis port=input_1,layer9_out 
```

vloncar, 2020-12-11:
`io_serial` isn't really supported anymore and will be removed in the next release. Even if it goes past conversion, there are other issues. If you want to use axis interface, you should consider `io_stream` from #220. It will be in master branch very soon.

jmduarte, 2021-09-12:
`io_serial` is now deprecated, and `io_stream` is recommended for this use case.

closed 2021-09-12 by jmduarte