fastmachinelearning/hls4ml #262: vivado_hls PATH

mechlya, 2020-12-14:
hello 
How do I make  vivado_hls  on PATH

Exception: Vivado HLS installation not found. Make sure "vivado_hls" is on PATH.
****

vloncar, 2020-12-14:
Hi @mechlya, you need to add the directory where Vivado is installed, for example `/tools/Xilinx/Vivado/2020.1/bin` to your `PATH` variable. See for example [here](https://linuxize.com/post/how-to-add-directory-to-path-in-linux/) how to do it.

closed 2020-12-14 by mechlya

Joash09, 2020-12-19:
Hi. I am attempting to use HLS4ML within the Windows Subsystem for Linux (WSL). 

The WSL allows me to use Windows apps within Linux. In this case I would like to run vivado_hls (which is installed in windows) from my Jupyter notebook and Conda environment which is running in the Linux subsystem. 

In order to run vivado_hls from the subsystem, I had to create the following alias in my WSL .bashrc. The hls is not an executable and you have to run the associated .bat file. 

`alias vivado_hls=cmd.exe /C \"..\\Xilinx\\Vivado\\2019.2\\bin\\vivado_hls.bat\"`

Therefore just running `vivado_hls` in the wsl terminal would open the application. 

My issue then is that line 542 of hls_model.py which uses os.system() to check if the tool can open vivado_hls however "alias" commands do not work in os.system(). I believe if we replace this line with [subprocess module](https://docs.python.org/3/library/subprocess.html) then I'd be able to carry on with my current dev environment. Is this possible? Please advise if this the correct channel for these discussions etc. This is my first time engaging with an open source project.

vloncar, 2020-12-21:
Hi @Joash09,
Xilinx doesn't support using Vivado through WSL so we would be reluctant to introduce workarounds in hls4ml to support this. You can make the change yourself and install that version. For example, clone the git repo to your work space, navigate to that directory and install with `pip install -e .` (note the dot at the end). Now you can make the changes and they will be reflected in the installed version. It should also be easy to keep the fork up-to-date, since we don't make many changes in that part of the code.