fastmachinelearning/hls4ml #264: cannot open shared object file: No such file or directory

mechlya, 2020-12-14:
Hello!

whene I compile the model I get this error 
 this is my first time using hls4mland I wonder if I have something setup wrong  !

hls_model.compile()
Writing HLS project
Done
build_lib.sh: line 14: g++: command not found
build_lib.sh: line 15: g++: command not found
build_lib.sh: line 16: g++: command not found
Traceback (most recent call last):

  File "<ipython-input-20-20374fcfe547>", line 1, in <module>
    hls_model.compile()

  File "/home/ahmed/anaconda3/lib/python3.8/site-packages/hls4ml/model/hls_model.py", line 402, in compile
    self._top_function_lib = ctypes.cdll.LoadLibrary(lib_name)

  File "/home/ahmed/anaconda3/lib/python3.8/ctypes/__init__.py", line 459, in LoadLibrary
    return self._dlltype(name)

  File "/home/ahmed/anaconda3/lib/python3.8/ctypes/__init__.py", line 381, in __init__
    self._handle = _dlopen(self._name, mode)

OSError: firmware/myproject-51a9dCAc.so: cannot open shared object file: No such file or directory

vloncar, 2020-12-14:
You need to install the GCC compiler on your system. If this is Ubuntu or Debian Linux, you can do it with `sudo apt install build-essential`. If on CentOS, you can do `sudo yum group install "Development Tools"`

mechlya, 2020-12-15:
After the completion I get this message
I don't know what it's mean

[image]

vloncar, 2020-12-15:
Your model is too large, so you ran out of memory during synthesis. You need to use `io_stream` instead of `io_parallel` to have any chance of success.

ribbles, 2026-09-03:
5y later and this didn't fix the issue :(