fastmachinelearning/hls4ml #266: INFO: [Common 17-206] Exiting vivado_hls

mechlya, 2020-12-17:
hello  

whne I build the project I get this error 

what's the prblem here !!

hls_model.build()

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
  **** IP Build 2548770 on Fri May 24 18:01:18 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source /opt/Xilinx/Vivado/2019.1/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2019.1/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'ahmed' on host 'ahmed-B75H2-AM2' (Linux_x86_64 version 5.4.0-58-generic) on Thu Dec 17 12:54:41 CET 2020
INFO: [HLS 200-10] On os Ubuntu 18.04.5 LTS
INFO: [HLS 200-10] In directory '/home/ahmed/download/test3/prj'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-10] Opening project '/home/ahmed/download/test3/prj/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Opening solution '/home/ahmed/download/test3/prj/myproject_prj/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [HLS 200-10] Setting target device to 'xcku115-flvb2104-2-i'
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../myproject_test.cpp in debug mode
   Compiling ../../../../firmware/myproject.cpp in debug mode
   Generating csim.exe
INFO: Unable to open input/predictions file, using default input.
@E Simulation failed: SIGSEGV.
ERROR: [SIM 211-100] CSim failed with errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Thu Dec 17 12:54:51 2020...

vloncar, 2020-12-17:
Segmentation fault. Impossible to say more without the model and the steps you took.

mechlya, 2020-12-17:
I just followed this tutorial 

https://github.com/fastmachinelearning/hls4ml-tutorial/blob/master/part1_getting_started.ipynb

this is the model architecture 

Interpreting Model
Topology:
Layer name: input_2, layer type: InputLayer, current shape: [[None, 128, 64, 3]]
Layer name: conv0, layer type: Conv2D, current shape: [[None, 128, 64, 3]]
Layer name: conv1, layer type: Conv2D, current shape: [[None, 128, 64, 32]]
Layer name: max_pooling2d_2, layer type: MaxPooling2D, current shape: [[None, 128, 64, 64]]
Layer name: conv3, layer type: Conv2D, current shape: [[None, 64, 32, 64]]
Layer name: max_pooling2d_3, layer type: MaxPooling2D, current shape: [[None, 64, 32, 128]]
Layer name: fc2, layer type: Dense, current shape: [[None, 65536]]
Creating HLS model

vloncar, 2020-12-17:
This likely won't work with hls4ml now. The model is too large.

rajeshpandit107, 2020-12-19:
Does hls4ml has limit on architecture size which it can synthesize?

vloncar, 2020-12-19:
hls4ml (currently) stores weights on the FPGA to achieve low latency. It is therefore limited by the resources available on the chip which varies from one product family to another.

closed 2022-06-19 by jmduarte