fastmachinelearning/hls4ml #269: ONNX converter issue

jmitrevs, 2021-01-06:
When running:

```
bash convert-onnx-models.sh -x -d test_prj -f onnx-models.txt
```

looking in particular at the `three_layer_keras` (though it's not unique), in this part of `onnx_to_hls`:

```python
        if layer['class_name'] == 'Dense':
            current_shape = get_input_shape(model, operation)
            layer['n_in'] = next((x.type.tensor_type.shape.dim[-1].dim_value for x in model.graph.input if x.name == operation.input[0]), None)
            layer['n_out'] = next((x.type.tensor_type.shape.dim[-1].dim_value for x in model.graph.value_info if x.name == operation.output[0]), None)
            tran_weight = get_onnx_attribute(operation, 'transB', 0)
            reader.add_input(layer['name'], operation.input, tran_weight)
```
`layer['n_out']` winds up with the value of None. Some debug output I made while looking at the Gemm5 layer shows:

```
current_shape=[1, 32]
operation.input=['fc2_relu_Relu_0', 'W2', 'B2']
list((x.name, x.type.tensor_type.shape.dim[-1].dim_value) for x in model.graph.input)=[('input_1_0', 16), ('W', 64), ('B', 64), ('W1', 32), ('B1', 32), ('W2', 32), ('B2', 32), ('W3', 5), ('B3', 5)]
Layer name: Gemm5, layer type: Dense, current shape: [1, 32]
```

In particular, note that `x.name == operation.input[0]` is always False when setting `layer['n_in']`. I am not sure what is the best solution for `layer['n_in']`. Isn't it just current_shape in a sense?

jmitrevs, 2021-01-08:
I also noticed differences in the shapes between pytorch-onnx and keras-onnx. In particular get_input_shape returns different shapes, and the weights matrices seem to be transposed (e.g. Keras produces a [16,64] matrix while pytorch [64,16]). 

I can probably take fixing these problems as my task to solve.

jmitrevs, 2021-01-08:
I will go ahead and make a PR for the keras-onnx problem and leave pytorch-onnx for a different time.

vloncar, 2021-01-08:
Hi Jovan,

Thanks a lot for looking into this. The ONNX converter is a bit outdated now and should indeed be updated. Are you planning on updating it completely, so it matches the design of Keras converter (which is also pursued in PyTorch PR)?

Duchstf, 2021-01-08:
@jmitrevs Let me know how far you get with fixing the onnx converter. I'm actually switching gear to update the onnx converter as well now. 

@vloncar I'm [discussing](https://github.com/Xilinx/brevitas/issues/235#issuecomment-756702578) with Alessandro about Brevitas models. Currently it seems like it's the best way forward for translating Brevitas models, Pytorch Geometric, and other bigger scale Pytorch models. I'll provide an update in next week's meeting.

jmitrevs, 2021-01-08:
Hi,

@Duchstf I haven't done much actual work, so if you want to fix it in a unified way, you can go ahead. But I am willing to spend time trying to fix it if you have lots of work on your hands.

Duchstf, 2021-01-08:
Thanks, actually can you send me the model you're trying to convert? would be good to test different models.

jmitrevs, 2021-01-08:
I was just doing the models in the test directory, running `bash convert-onnx-models.sh -x -f  onnx-models.txt`.

jmitrevs, 2021-01-09:
By the way, a minimal fix that makes the ONNX test pass in the quartus branch (though the error is not quartus-specific) is:

```diff
(fastml) jovans-mac:test jmitrevs$ git diff
diff --git a/hls4ml/converters/onnx_to_hls.py b/hls4ml/converters/onnx_to_hls.py
index c1356a1..15b851f 100644
--- a/hls4ml/converters/onnx_to_hls.py
+++ b/hls4ml/converters/onnx_to_hls.py
@@ -222,7 +222,8 @@ def onnx_to_hls(yamlConfig):
         #(We take it from the weights to avoid dealing with InputLayer and Flatten details)
         if layer['class_name'] == 'Dense':
             current_shape = get_input_shape(model, operation)
-            layer['n_in'] = next((x.type.tensor_type.shape.dim[-1].dim_value for x in model.graph.input if x.name == operation.input[0]), None)
+            layer['n_in'] = next((x.type.tensor_type.shape.dim[-1].dim_value for x in model.graph.value_info if x.name == operation.input[0]),
+                                 next((x.type.tensor_type.shape.dim[-1].dim_value for x in model.graph.input if x.name == operation.input[0]), None))
             layer['n_out'] = next((x.type.tensor_type.shape.dim[-1].dim_value for x in model.graph.value_info if x.name == operation.output[0]), None)
             tran_weight = get_onnx_attribute(operation, 'transB', 0)
             reader.add_input(layer['name'], operation.input, tran_weight)
```
This logic also works for the pytorch-onnx models, though those have other issues (for example, `get_input_shape` returns the wrong value). And truthfully not much of the converter is tested in this test, so there could well be other problems.

I didn't quite understand if you want me to take a look at updating the ONNX converter more generally or if you are doing it.

Duchstf, 2021-01-10:
@jmitrevs Hello, I'm working on updating the ONNX more generally, but if you want to make a quick PR to fix your problems go ahead.

jmitrevs, 2021-01-11:
I made PR #272 with this small fix.

jmduarte, 2021-09-10:
Should be addressed by #285

closed 2021-09-10 by jmduarte