fastmachinelearning/hls4ml #270: WEIGHTS_DIR redefinition

KOVI89alipes, 2021-01-07:
`WEIGHTS_DIR` is hard-coded in
https://github.com/fastmachinelearning/hls4ml/blob/f31b52fca2a85deb8d1b91d637636a365151194f/hls4ml/templates/vivado/nnet_utils/nnet_helpers.h#L35
And it requires copying or symlinking `weights` folder into the build folder whenever the model function is compiled from the outside of HLS4ML project. 

Possible  fix: 
```
#ifndef WEIGHTS_DIR
#define WEIGHTS_DIR "weights"
#endif
```
In this case WEIGHTS_DIR can be redefined during compilation

Another option - do not store TXT files, but dump weights into `includes`

vloncar, 2021-01-08:
Can you expand a little bit on your workflow? We can make the change you propose, but note that if you generate weights yourself from outside of hls4ml, you need both the `.txt` files and `.h` includes. Text files are used only during C simulation to speed up compilation. During synthesis, we have to use the includes.

KOVI89alipes, 2021-01-11:
Hi @vloncar , it's clear about synthesis, but I'm talking about the situation when one needs to call the top C++ function as normal function.

Let's here's my file structure and I wish to use `my_fancy_core` function from my `host_app`
```
work/ip_cores/my_fancy_core/
			my_fancy_core_test.cpp
			firmware/
				my_fancy_core.cpp
				my_fancy_core.h
				weights/*.txt << Txt files here

work/host_app/
	main.cpp
```

And here's my `main.cpp` program calls  `my_fancy_core` function

```
===== main.cpp ====
#include "firmware/my_fancy_core_test.h"  << Path work/ip_cores/my_fancy_core/ was added during compilation
int main(){
     // Layes assignment goes here
    my_fancy_core_test(input_1,layer9_out,size_in1,size_out1);	
}

```

main.cpp can be compiled with no errors, but when I call the binary, it fails with
https://github.com/fastmachinelearning/hls4ml/blob/f31b52fca2a85deb8d1b91d637636a365151194f/hls4ml/templates/vivado/nnet_utils/nnet_helpers.h#L43
```
    if (infile.fail()) {
        std::cerr << "ERROR: file " << std::string(fname) << " does not exist" << std::endl;
        exit(1);
    }
```
Because directory `weights` obviously doesn't exist in the `work\host_app` dir and I can not redefine `weights_dir` from the compilation flow.

So I believe it makes sense to let the user an option to redefine  `WEIGHTS_DIR`  from the compilation flow.

vloncar, 2021-01-15:
@KOVI89alipes That's reasonable. I created #274 , sorry for the delay.

closed 2023-03-16 by jmduarte