fastmachinelearning/hls4ml #275: Resources for HLS codes in nnet_utils.

ninfueng, 2021-01-15:
Hello,

I try to understand HLS codes in `nnet_utils`. However, I quite confuse the terminology and design within them.
For example, there are four header files for 2d convolutional layer: `nnet_conv2d.h`, `nnet_conv2d_latency.h`, `nnet_conv2d_resource.h`, and `nnet_conv2d_stream.h`. I did not understand what is use case of `nnet_conv2d_latency.h`, and `nnet_conv2d_resource.h`.

Therefore, if there is any, I would like to ask for resources or documentation related to these HLS codes to learn. 
So far what I found are related to Python and overview concepts.

Thank you.

vloncar, 2021-01-15:
Hello,

Multiple implementations correspond to the different `Strategy` and `IOType` [configuration](https://fastmachinelearning.org/hls4ml/api/CONFIGURATION.html) parameters. You can read about `nnet_conv2d_stream.h` [here](https://arxiv.org/abs/2101.05108). The `nnet_conv2d_latency.h` is an implementation of direct convolution, while `nnet_conv2d_resource.h` is based on im2col, but working on one column of im2col matrix at a time. `conv2d_latency` and `conv2d_resource` may get the axe soon though.

Hope that helps.

Cheers

ninfueng, 2021-01-16:
Hello @vloncar,

Thank you for the link and paper. I will read them.

closed 2021-01-16 by ninfueng