fastmachinelearning/hls4ml #276: OSError: firmware/myproject-73DAB14B.so: cannot open shared object file: No such file or directory

Duchstf, 2021-01-19:
I'm rewriting the ONNX converter and got the conversion to work. However, when I try compiling the hls model,I got this error:

```---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-4-20374fcfe547> in <module>
----> 1 hls_model.compile()

~/pytorch_dev/hls4ml/hls4ml/model/hls_model.py in compile(self)
    404                 dlclose_func.restype = ctypes.c_int
    405                 dlclose_func(self._top_function_lib._handle)
--> 406             self._top_function_lib = ctypes.cdll.LoadLibrary(lib_name)
    407         finally:
    408             os.chdir(curr_dir)

~/miniconda3/envs/pytorch/lib/python3.8/ctypes/__init__.py in LoadLibrary(self, name)
    457 
    458     def LoadLibrary(self, name):
--> 459         return self._dlltype(name)
    460 
    461 cdll = LibraryLoader(CDLL)

~/miniconda3/envs/pytorch/lib/python3.8/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error, winmode)
    379 
    380         if handle is None:
--> 381             self._handle = _dlopen(self._name, mode)
    382         else:
    383             self._handle = handle

OSError: firmware/myproject-73DAB14B.so: cannot open shared object file: No such file or directory
```

I suspect that this is because I'm working on a fresh conda environment that I created on correlator2, so some libraries might not be set up appropriately ...

Duchstf, 2021-01-19:
Never mind, it's just an error caused by not processing the layer's class name appropriately.

closed 2021-01-19 by Duchstf

thesps, 2021-05-11:
Generally, this error means that the compilation of the C++ failed, so the shared library doesn't exist. You could try running the model 'build' to see the exact error in the Vivado log files.

vinayak19th, 2021-05-11:
Yeah. there is a build error with no matches for operator "=". Seems to be an issue with the way concat was implemented and how it works with conv layers