fastmachinelearning/hls4ml #279: Uses of granularity

rajeshpandit107, 2021-01-28:
Can someone please explain the use of granularity while defining the model configuration?

vloncar, 2021-01-28:
Configuration is meant to be tweaked. You decide how deep you want to go with tweaking. Maybe you just want to specify one precision for the whole model, or maybe you want to specify that one particular layer is going to use specific precision to store intermediate multiplication values. Granularity helps with making that initial configuration.

rajeshpandit107, 2021-02-13:
Can We synthesize a single layer of a given model(for example first convolution layer of Lenet 5 architecture)  using 'name' or 'type' granularity or any other method?

vloncar, 2021-02-15:
hls4ml synthesizes whole models, if you only want to synthesize one layer, create a model with only one layer.

rajeshpandit107, 2021-02-15:
Thanks for that quick reply.
How much time did it take to synthesize a large CNN model in version 0.5.0 version of hls4ml(The CNN model that has been given as reference in the second published paper of hls4ml) ?
[image]

thaarres, 2021-02-15:
Hi @rajeshpandit107! The model takes roughly 45-90 minutes to synthesise, depending on your CPU.

rajeshpandit107, 2021-03-01:
hi @thaarres , I am trying to regenerate a similar cnn model, but my synthesis neither completes nor gives any error, it gets stuck for 4-5 days. Could you please help me with the way to define synthesis configuration ?

rajeshpandit107, 2021-03-01:
I am using a 3 GHz intel Xeon CPU with a ram of 32GB

thaarres, 2021-03-01:
Hi @rajeshpandit107. This sounds strange. Could you provide further details on what your model is/what your hls4ml configuration is set to? You can find an example configuration at

[https://github.com/thaarres/hls4ml_cnns/blob/master/doScans.py#L61 ](url)

rajeshpandit107, 2021-03-10:
Hi @thaarres , I tried re-synthesizing the model, I could not succeed. Could you please see the configuration file and point out the incorrect configuration paramters
[vivado_hls.log](https://github.com/fastmachinelearning/hls4ml/files/6114872/vivado_hls.log)
[model.py.log](https://github.com/fastmachinelearning/hls4ml/files/6114878/model.py.log)

thaarres, 2021-03-23:
Hi @rajeshpandit107 !
You need to set `cfg['IOType']     = 'io_stream'`.
See example at
https://github.com/thaarres/hls4ml_cnns/blob/master/doScans.py#L49

closed 2023-03-16 by jmduarte