fastmachinelearning/hls4ml #287: Fail to build the example model

JunningFan, 2021-02-26:
Hi!
I'm learning to convert models to a Vivado hls project. I followed the setup section of the hls4ml documentation and tried to convert the model defined in example-models/keras-config.yml. However, the hls4ml build -p my-hls-test -a command failed as it seems the CSIM reported a compilation error. Can someone help me solving the issue? Thanks.

hls4ml convert -c keras-config.yml
/opt/anaconda/bin/hls4ml:171: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if found is not 0:
2021-02-27 03:23:09.749297: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.11.0
Loading configuration from keras-config.yml
Interpreting Model
Topology:
Layer name: input_1, layer type: InputLayer, current shape: [[None, 16]]
Layer name: fc1_relu, layer type: Dense, current shape: [[None, 16]]
Layer name: fc2_relu, layer type: Dense, current shape: [[None, 64]]
Layer name: fc3_relu, layer type: Dense, current shape: [[None, 32]]
Layer name: output_softmax, layer type: Dense, current shape: [[None, 32]]
Creating HLS model
Writing HLS project
Done
haoxi1999@haoxi1999-PC:~/Documents/hls4ml/example-models$ hls4ml build -p my-hls-test -a
/opt/anaconda/bin/hls4ml:171: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if found is not 0:
2021-02-27 03:23:16.946494: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.11.0
Loading configuration from my-hls-test/hls4ml_config.yml

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.2 (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:02:38 MDT 2018
  **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source /opt/Xilinx/Vivado/2018.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2018.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'haoxi1999' on host 'haoxi1999-PC' (Linux_x86_64 version 5.8.0-44-generic) on Sat Feb 27 03:23:18 AEDT 2021
INFO: [HLS 200-10] On os Ubuntu 20.04.2 LTS
INFO: [HLS 200-10] In directory '/home/haoxi1999/Documents/hls4ml/example-models/my-hls-test'
INFO: [HLS 200-10] Opening project '/home/haoxi1999/Documents/hls4ml/example-models/my-hls-test/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Opening solution '/home/haoxi1999/Documents/hls4ml/example-models/my-hls-test/myproject_prj/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-10] Setting target device to 'xc7a200tsbg484-1'
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Sat Feb 27 03:23:18 2021...

vloncar, 2021-03-01:
This could be related to the Vivado installation. You are running a version 2018.2 on Ubuntu 20.04, a combination that Xilinx doesn't support. Just to be sure, can you share the generated project to make sure nothing was wrong with the code-generation?

JunningFan, 2021-03-02:
Hi vloncar

Here's the generated project attacjed

[proj.zip](https://drive.google.com/file/d/1gmY39LOik-o7KuYI-VpV3O7M_FBG-hrU/view?usp=sharing)

closed 2023-03-16 by jmduarte