fastmachinelearning/hls4ml #289: Pre-synthesis failed when building a network with QKeras Layers

JunningFan, 2021-03-02:
Hi, 

I'm trying yo use hls4ml to translate a network model with QKeras layers. Sadly, the generated HLS project failed to synthesis in vivado_hls tool. The log, network model, YAML config, and the generated HLS is attached. Any help is appreciated.
 
[model.zip](https://github.com/fastmachinelearning/hls4ml/files/6066033/model.zip)
[project.zip](https://drive.google.com/file/d/178uDMqAqutV3avYF6jP0veiHbnimGDsM/view?usp=sharing)

Environment:
    Windows 10 20H2
    Vivado and Vivado HLS 2018.2
    HLS4ML 0.5.0b0

Log:
SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if found is not 0:
2021-03-02 13:23:17.163205: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library cudart64_110.dll
Loading configuration from .\my-hls-test/hls4ml_config.yml

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.2 (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:03:12 MDT 2018
  **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source D:/Xilinx/Vivado/2018.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running 'D:/Xilinx/Vivado/2018.2/bin/unwrapped/win64.o/vivado_hls.exe'
INFO: [HLS 200-10] For user 'haoxi' on host 'haoxi-pc' (Windows NT_amd64 version 6.2) on Tue Mar 02 13:23:20 +1100 2021
INFO: [HLS 200-10] In directory 'E:/DLProjects/hls4ml/my-hls-test'
INFO: [HLS 200-10] Creating and opening project 'E:/DLProjects/hls4ml/my-hls-test/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Creating and opening solution 'E:/DLProjects/hls4ml/my-hls-test/myproject_prj/solution1'.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg484-1'
INFO: [SYN 201-201] Setting up clock 'default' with a period of 20ns.
INFO: [Common 17-206] Exiting vivado_hls at Tue Mar  2 13:23:20 2021...
PS E:\DLProjects\hls4ml> python C:\Users\haoxi\anaconda3\Scripts\hls4ml build -p .\my-hls-test -a
C:\Users\haoxi\anaconda3\Scripts\hls4ml:171: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if found is not 0:
2021-03-02 13:23:24.678717: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library cudart64_110.dll
Loading configuration from .\my-hls-test/hls4ml_config.yml

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.2 (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:03:12 MDT 2018
  **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source D:/Xilinx/Vivado/2018.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running 'D:/Xilinx/Vivado/2018.2/bin/unwrapped/win64.o/vivado_hls.exe'
INFO: [HLS 200-10] For user 'haoxi' on host 'haoxi-pc' (Windows NT_amd64 version 6.2) on Tue Mar 02 13:23:27 +1100 2021
INFO: [HLS 200-10] In directory 'E:/DLProjects/hls4ml/my-hls-test'
INFO: [HLS 200-10] Opening project 'E:/DLProjects/hls4ml/my-hls-test/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Opening solution 'E:/DLProjects/hls4ml/my-hls-test/myproject_prj/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 20ns.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg484-1'
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-1161] The maximum of name length is set into 60.
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../myproject_test.cpp in debug mode
   Compiling ../../../../firmware/myproject.cpp in debug mode
[... 203 lines cut ...]
INFO: [XFORM 203-501] Unrolling loop 'ReduceLeft' (firmware/nnet_utils/nnet_common.h:56) in function 'nnet::reduce<ap_fixed<18, 8, (ap_q_mode)5, (ap_o_mode)3, 0>, 8, nnet::Op_add<ap_fixed<18, 8, (ap_q_mode)5, (ap_o_mode)3, 0> > >' completely.
INFO: [XFORM 203-501] Unrolling loop 'ReduceRight' (firmware/nnet_utils/nnet_common.h:59) in function 'nnet::reduce<ap_fixed<18, 8, (ap_q_mode)5, (ap_o_mode)3, 0>, 8, nnet::Op_add<ap_fixed<18, 8, (ap_q_mode)5, (ap_o_mode)3, 0> > >' completely.
INFO: [XFORM 203-501] Unrolling loop 'ReduceLeft' (firmware/nnet_utils/nnet_common.h:56) in function 'nnet::reduce<ap_fixed<18, 8, (ap_q_mode)5, (ap_o_mode)3, 0>, 4, nnet::Op_add<ap_fixed<18, 8, (ap_q_mode)5, (ap_o_mode)3, 0> > >' completely.
INFO: [XFORM 203-501] Unrolling loop 'ReduceRight' (firmware/nnet_utils/nnet_common.h:59) in function 'nnet::reduce<ap_fixed<18, 8, (ap_q_mode)5, (ap_o_mode)3, 0>, 4, nnet::Op_add<ap_fixed<18, 8, (ap_q_mode)5, (ap_o_mode)3, 0> > >' completely.
INFO: [XFORM 203-501] Unrolling loop 'Loop-1' (firmware/nnet_utils/nnet_activation.h:76) in function 'nnet::relu<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, relu_config13>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Result' (firmware/nnet_utils/nnet_batchnorm.h:83) in function 'nnet::normalize<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config18>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense_latency.h:85) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config11>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Product2' (firmware/nnet_utils/nnet_dense_latency.h:90) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config11>' completely.
INFO: [XFORM 203-501] Unrolling loop 'ResetAccum' (firmware/nnet_utils/nnet_dense_latency.h:101) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config11>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Accum1' (firmware/nnet_utils/nnet_dense_latency.h:109) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config11>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Accum2' (firmware/nnet_utils/nnet_dense_latency.h:113) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config11>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Result' (firmware/nnet_utils/nnet_dense_latency.h:120) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config11>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Loop-1' (firmware/nnet_utils/nnet_activation.h:76) in function 'nnet::relu<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, relu_config10>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Result' (firmware/nnet_utils/nnet_batchnorm.h:83) in function 'nnet::normalize<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config17>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense_latency.h:85) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config8>' completely.
ERROR: [XFORM 203-504] Stop unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense_latency.h:85) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config8>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Tue Mar  2 13:57:15 2021...

thesps, 2021-03-02:
For a CNN model you should set `IOType: io_stream` in your configuration. Then, if you still see such errors, or the final resources are too large, you should set `Strategy: Resource` in the `HLSConfig` section. And finally, try increasing the `ReuseFactor`.

I can't be sure, but it also looks like you may have `ap_fixed<16,6>` data types, while you mentioned using QKeras layers in the title. To derive the data types portion of the configuration for a QKeras model, you should generate your config template like:
`config = hls4ml.utils.config_from_keras_model(model, granularity='name')`
See the '[quantization](https://github.com/fastmachinelearning/hls4ml-tutorial/blob/master/part4_quantization.ipynb)' part of the hls4ml tutorial for more details, or try it online:

[image]

JunningFan, 2021-03-04:
> For a CNN model you should set `IOType: io_stream` in your configuration. Then, if you still see such errors, or the final resources are too large, you should set `Strategy: Resource` in the [...]

Hi thesps,

Thanks for helping. Specifying Resource does resolve the [XFORM 203-504] error for the dense layer. Unfortunately, I have noticed that specifying the Resource strategy changes the output of the network. On a CNN trained to classify MNIST images, the simulated accuracy of the hls_model with latency is expected (.97), however, the simulated accuracy of the model with Resource strategy seems to be broken(0.127).  
[image]
[image]
The generated config file with resource strategy is attached:

-----------------------------------
Model
  Precision:         ap_fixed<10,3>
  ReuseFactor:       16
  Strategy:          Resource
LayerName
  input_1
    Precision
      result:        ap_ufixed<8,1>
  conv2d_0_m
    Precision
      weight:        ap_fixed<9,1>
      bias:          ap_fixed<9,1>
    ReuseFactor:     16
  act0_m
    Precision
      result:        ap_ufixed<8,1>
    ReuseFactor:     16
  max_pooling2d
    Precision:       ap_fixed<16,6>
  conv2d_1_m
    Precision
      weight:        ap_fixed<9,1>
      bias:          ap_fixed<9,1>
    ReuseFactor:     16
  act1_m
    Precision
      result:        ap_fixed<9,1>
    ReuseFactor:     16
  fc_0_m
    Precision
      weight:        ap_fixed<9,1>
      bias:          ap_fixed<9,1>
    ReuseFactor:     16
  softmax
    Precision:       ap_fixed<16,6>
    ReuseFactor:     16
    table_size:      1024
    exp_table_t:     ap_fixed<18,8,AP_RND,AP_SAT>
    inv_table_t:     ap_fixed<18,8,AP_RND,AP_SAT>

waldenou, 2021-03-07:
Hi,

I have encountered a similar issue with my CNN model. The model was trained using MNIST data. When I use `Strategy: Latency + IOType: io_stream`, the hls4ml model output accuracy is near the keras model accuracy, but the resource usage is undesirable. When I use `Strategy: Resource + IOType: io_stream` the ouput accuracy is only around 10%, and the converted model seems to predict one class more often than the others.

Any help is appreciated. Thanks.

vloncar, 2021-03-08:
@JunningFan @waldenou this is solved in #299

closed 2026-10-26 by github-actions[bot]