fastmachinelearning/hls4ml #295: Vivado/Vitis version

SerenaC94, 2021-03-04:
I am having trouble generating basic projects (for example the ones in example-prjs) and I think it could be related to my Xilinx software installation.

Trivial problem & workaround: hls_model.build() does not find the vivado_hls executable because in version 20.2 it is called vitis_hls, so I tried

`vitis_hls -f build_prj.tcl `

...but then I get a bunch of errors, and they are different for each one of the folders in example-prjs. (Are the models not up to date perhaps?)

What is the latest version of Vivado HLS supported by hls4ml?

SerenaC94, 2021-03-04:
Here is some additional information.

I have naively modified `hls4ml/model/hls_model.py` so that it looks for and launches vitis_hls. Then I have a simple script that follows the guide in the homepage:

```
import hls4ml

config = hls4ml.utils.fetch_example_model('KERAS_3layer.json')

hls_model = hls4ml.converters.keras_to_hls(config)

hls_model.build()

```

And I get the following errors (with a lot of other info messages in the middle):

```
ERROR: [HLS 200-642] The 'config_array_partition -maximum_size' command is not supported.

***** C/RTL SYNTHESIS *****
INFO: [HLS 200-1510] Running: csynth_design 
INFO: [HLS 200-111] Finished File checks and directory preparation: CPU user time: 0.02 seconds. CPU system time: 0 seconds. Elapsed time: 0.02 seconds; current allocated memory: 191.006 MB.
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ... 
ERROR: [HLS 214-125] no member named 'vector' in namespace 'std': /hls4ml/my-hls-test/firmware/nnet_utils/nnet_helpers.h:279
ERROR: [HLS 214-124] use of undeclared identifier 'src': /hls4ml/my-hls-test/firmware/nnet_utils/nnet_helpers.h:279
ERROR: [HLS 214-123] expected unqualified-id: /hls4ml/my-hls-test/firmware/nnet_utils/nnet_helpers.h:279
ERROR: [HLS 214-125] no member named 'vector' in namespace 'std': /hls4ml/my-hls-test/firmware/nnet_utils/nnet_helpers.h:286
ERROR: [HLS 214-124] use of undeclared identifier 'src': /hls4ml/my-hls-test/firmware/nnet_utils/nnet_helpers.h:286
ERROR: [HLS 214-124] use of undeclared identifier 'dst': /hls4ml/my-hls-test/firmware/nnet_utils/nnet_helpers.h:286
ERROR: [HLS 214-123] expected unqualified-id: /hls4ml/my-hls-test/firmware/nnet_utils/nnet_helpers.h:286
INFO: [HLS 200-111] Finished Command csynth_design CPU user time: 5.61 seconds. CPU system time: 0.38 seconds. Elapsed time: 5.99 seconds; current allocated memory: 193.079 MB.
Syntax check failed in clang-tidy
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

```

vloncar, 2021-03-04:
Vitis HLS and Vivado HLS are two different products, and we don't yet support Vitis HLS. You can use Vivado HLS 2020.1

SerenaC94, 2021-03-04:
Understood, thanks. I think that the README is somewhat misleading, because it contains a link redirecting to the main Xilinx Vitis webpage, and Vivado HLS does not exist anymore in the latest version of the full suite.

closed 2021-03-04 by SerenaC94

thesps, 2021-03-04:
It's a good point. We're in the process of changing some documentation for the next release, and we will make that part more clear.