fastmachinelearning/hls4ml #297: Remove alternative C++ operator

KOVI89alipes, 2021-03-08:
Hi,
One of the HLS template files uses old-style, alternative  logical C++ operator `and` and `not`

https://github.com/fastmachinelearning/hls4ml/blob/c3509cfeb5f80054e01dead612392a9411ebfbe2/hls4ml/templates/vivado/nnet_utils/nnet_mult.h#L93

```
template<class data_T, class res_T, typename CONFIG_T>
inline typename std::enable_if<std::is_same<data_T, ap_uint<1>>::value
        and std::is_same<typename CONFIG_T::weight_t, ap_uint<1>>::value, ap_int<nnet::ceillog2(CONFIG_T::n_in) + 2>>::type
cast(typename CONFIG_T::accum_t x){
  return (ap_int<nnet::ceillog2(CONFIG_T::n_in) + 2>) (x - CONFIG_T::n_in / 2) * 2;
}

template<class data_T, class res_T, typename CONFIG_T>
inline typename std::enable_if<(not std::is_same<data_T, ap_uint<1>>::value), res_T>::type
cast(typename CONFIG_T::accum_t x){
  return (res_T) x;
}
```

These are alternative tokens
https://en.cppreference.com/w/cpp/language/operator_alternative

And they are not supported by the MSVC compiler
https://stackoverflow.com/questions/24414124/why-does-vs-not-define-the-alternative-tokens-for-logical-operators

It prevents HLS4ML kernel code to be compiled in Windows with MSVC and gives an error
```
firmware\nnet_utils\nnet_mult.h(88): error C2146: syntax error: missing '>' before identifier 'and' 
```

Suggestion: replace alternative logical operators with standard ones `and` -> `&&`, `not`->`~`

```
template<class data_T, class res_T, typename CONFIG_T>
inline typename std::enable_if<std::is_same<data_T, ap_uint<1>>::value
        && std::is_same<typename CONFIG_T::weight_t, ap_uint<1>>::value, ap_int<nnet::ceillog2(CONFIG_T::n_in) + 2>>::type
cast(typename CONFIG_T::accum_t x){
  return (ap_int<nnet::ceillog2(CONFIG_T::n_in) + 2>) (x - CONFIG_T::n_in / 2) * 2;
}

template<class data_T, class res_T, typename CONFIG_T>
inline typename std::enable_if<(! std::is_same<data_T, ap_uint<1>>::value), res_T>::type
cast(typename CONFIG_T::accum_t x){
  return (res_T) x;
}
```
Code with these changes can be compiled in MSVC and is still valid in GCC, Vivado HLS

Why to bother - we are using HLS4ML kernels in Windows and it's nice to have bittrue model of the kernel for verification purposes.

thesps, 2021-03-08:
I think we could now remove these `enable_if` constructions altogether, or at least simplify the logic a lot. The logic to decide which is the appropriate `cast` function is essentially already now done in the Python to decide which `Product` to use.

Logically, the `cast` function containing the `ceillog2` should be used when the `both_binary` class is used, while the second (general) one should be used for all other cases.

KOVI89alipes, 2021-03-09:
These two operators and this issue are the only things that prevent ANN HLS code to be compiled in Windows.
https://github.com/fastmachinelearning/hls4ml/issues/300

For our team, it makes a lof sense to have this feature as we use FPGAs with HLS4ML kernels both in Linux and Windows environment

KOVI89alipes, 2021-04-08:
@thesps for some reason, we didn't push this fix to the master branch of the origin

Could you please have a look at it
https://github.com/fastmachinelearning/hls4ml/pull/318

closed 2026-10-26 by github-actions[bot]