fastmachinelearning/hls4ml #303: Load weights in a compile time

KOVI89alipes, 2021-03-11:
Hi,
Just out of curiosity, what was the motivation to load ANN weights in a runtime in the simulation mode, not in a compile-time, like in Synthesis?

The drawbacks of the runtime loading
- `WEIGHTS_DIR` should be defined in a compile-time
- Every time you call `hls4ml_ann_name(input_1_ap, layer9_out_ap, const_size_in_1, const_size_out_1);` you have to read N txt files and transform string to integers

We have a local fix that replaces in  `w#.h` files

```
#ifndef __INITIALIZE_WEIGHTS__ // Instead of #ifndef __SYNTHETHIS__

model_default_t b2[64];
#else
model_default_t b2[64] = {-0.6503423452, .... };
#endif
```
and define `__INITIALIZE_WEIGHTS__` in  `parameters.h`  for synthethis

```
//hls-fpga-machine-learning insert weights
#ifdef  __SYNTHESIS__
#define __INITIALIZE_WEIGHTS__
#endif

#include "weights/w2.h"
```

Default behavior preserves the same
- In the HLS Synthesis weights are still compile-time defined
- In normal gcc/msvc compilation  weights are loaded in the runtime at each function call

But one can set "-D__INITIALIZE_WEIGHTS__" in the compilation and then weights will be compile-time defined.

Does it makes any sense for you? Shall we do a merge request or your don't really need it?

vloncar, 2021-03-11:
> Just out of curiosity, what was the motivation to load ANN weights in a runtime in the simulation mode, not in a compile-time, like in Synthesis?

We found that it increases the compile time and memory used. Since you will typically run C sim much more than synthesis, it adds up.

> * `WEIGHTS_DIR` should be defined in a compile-time

[You asked for this, remember?](https://github.com/fastmachinelearning/hls4ml/issues/270)

> * Every time you call `hls4ml_ann_name(input_1_ap, layer9_out_ap, const_size_in_1, const_size_out_1);` you have to read N txt files and transform string to integers

This [flag](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/templates/vivado/firmware/myproject.cpp#L31) guarantees that the file is only read once.

> Does it makes any sense for you? Shall we do a merge request or your don't really need it?

I'm not seeing the usefulness. Do you have a scenario where it doesn't work?

KOVI89alipes, 2021-03-12:
> You asked for this, remember?

Yes, I do :)
For the flag and reading during each call - my bad, I skipped it 

We  redefine `weights_dir` during the compilation, but the issue is still there
-  weights file shall be stored either next to the library/binary files and the path can be relative
-  or we have to make the path absolute. 
And txt files still must be delivered as a separate folder, not in as compiled `*.so/*.dll`

Compile-time reasoning makes a lot of sense, now I see your motivation
So the proposed mechanism preserves the default behavior and keeps compile-time low when you do co-simulation, but also adds an option to make weights compile-time defined, without the necessity to handle txt files when you deliver the IP core model as the library.

vloncar, 2021-03-13:
What is the big picture here, how are you integrating hls4ml? Perhaps we could schedule a meeting to map out any future changes required to accommodate cleaner integration of hls4ml into bigger projects or in different domains.

KOVI89alipes, 2021-03-15:
That's a brief explanation of out current system

- We have a common numerical framework, that can run computations on the different backends - GPU, CPU and now FPGAs.
- HLS4ML kernel with AXI interfaces communicates with OS (nix/win) through to Xilinx XMDA PCIE IP core and XDMA drivers.
- C++ backend code works on top of the XDMA driver, configurate and communicate with the core and gives general API in a form of `fpga_kernel.predict(X,y)`
- For the verification purpose we wish to have HLS C++ code as reference, bittrue model of the FPGA kernel and so we could call something like `kernel.predict_sw(X,y)` and compare it with real FPGA prediction. Yes, we are aware of C/RTL co-simulation stage as part of HLS4ML library, but we still need this stage to ensure data integrity on the system level. To make sure that we don't data racing in the PCIe or AXI infrastructure, that kernel properly configured, and that FPGA timing violation are not so critical  ( :)  )
- The drawback of keeping weights reading runtime procedure for us is that we have to deliver txt files with the binary, compiled SO/DLL  object. It would be nice to have an option to make them compile time defined. 

@vloncar It would be really nice, if we could have a meeting together, I'll send you an email

closed 2023-03-16 by jmduarte