fastmachinelearning/hls4ml #305: "No known conversion for arguments" in zeropad*d_cl functions

GiuseppeDiGuglielmo, 2021-03-20:
I attach a package that shows a compilation problem that we are encountering.

```
tar xvfz my-hls-test-baseline.tgz
cd my-hls-test-baseline
vivado_hls -f build_prj.tcl
```

Functions like _zeropad2d_cl_ "redirect" an hls::stream to another:
https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/templates/vivado/nnet_utils/nnet_padding_stream.h#L54

There are situations where the type of the input (_data_T_) and output (_res_T_) are different.
https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/templates/vivado/nnet_utils/nnet_padding_stream.h#L37-L41

For example, 
- the input comes from an activation function whose type is `nnet::array<ap_fixed<8, 4, (ap_q_mode)0u, (ap_o_mode)0u>, 8u>` 
- the output that goes to a convolutional layer is `nnet::array<ap_fixed<8, 4>, 8u>`

In this kind of situations, the compiler would complain
```
In file included from ../../../../firmware/parameters.h:19:0,
                 from ../../../../firmware/resnet_v1_eembc.cpp:22:
../../../../firmware/nnet_utils/nnet_padding_stream.h: In instantiation of ‘void nnet::zeropad2d_cl(hls::stream<srcType>&, hls::stream<dstType>&) [with data_T = nnet::array<ap_fixed<8, 4, (ap_q_mode)0u, (ap_o_mode)0u>, 16u>; res_T = nnet::array<ap_fixed<8, 4>, 16u>; CONFIG_T = config43]’:
../../../../firmware/resnet_v1_eembc.cpp:113:78:   required from here
../../../../firmware/nnet_utils/nnet_padding_stream.h:54:13: error: no matching function for call to ‘hls::stream<nnet::array<ap_fixed<8, 4>, 16u> >::write(nnet::array<ap_fixed<8, 4, (ap_q_mode)0u, (ap_o_mode)0u>, 16u>)’
             res.write(data.read());
             ^~~
In file included from ../../../../firmware/resnet_v1_eembc.h:25:0,
                 from ../../../../firmware/resnet_v1_eembc.cpp:21:
/tools/Xilinx/Vivado/2019.1/include/hls_stream.h:224:10: note: candidate: void hls::stream<__STREAM_T__>::write(const __STREAM_T__&) [with __STREAM_T__ = nnet::array<ap_fixed<8, 4>, 16u>]
     void write(const __STREAM_T__& tail) {
          ^~~~~
/tools/Xilinx/Vivado/2019.1/include/hls_stream.h:224:10: note:   no known conversion for argument 1 from ‘nnet::array<ap_fixed<8, 4, (ap_q_mode)0u, (ap_o_mode)0u>, 16u>’ to ‘const nnet::array<ap_fixed<8, 4>, 16u>&’
```
The problem may be associated with the `nnet::array` type which may not handle the assignment between different types:
https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/templates/vivado/nnet_utils/nnet_types.h#L24-L26

[my-hls-test-baseline.zip](https://github.com/fastmachinelearning/hls4ml/files/6174626/my-hls-test-baseline.zip)

vloncar, 2021-03-22:
Hi Giuseppe, thanks for the report. The issue you describe can be handled by explicit copying of data in that function. See 87ca3610e7e903f1ad802eb9bb78ec3c1f5a74d7 in [cnn_stream](https://github.com/vloncar/hls4ml/tree/cnn_stream) branch.

However, this is not the only issue here. Due to the limitation of graph manipulation routines (i.e., a bug) and the order in which optimizers are applied, the ZeroPadding layer gets inserted between another ZeroPadding layer and Conv instead of Activation and Conv. This will require some more thinking to properly solve.

GiuseppeDiGuglielmo, 2021-03-22:
Vladimir, thanks for having a look at this. The _explicit copy_ may work indeed. Hopefully with the unrolling that functions gets optimized away (no extra logic, just wires).

I am a little more concerned about the second issue you brought up. From what I see, the ZeroPadding seems correctly interleaved with the other layers, but I may have been looking at the wrong location:
```c++
    hls::stream<layer6_t> layer6_out("layer6_out");
    #pragma HLS STREAM variable=layer6_out depth=1024
    #pragma HLS STABLE variable=layer6_out
    nnet::conv_2d_cl<layer42_t, layer6_t, config6>(layer42_out, layer6_out, w6, b6); // conv2d_1

    hls::stream<layer9_t> layer9_out("layer9_out");
    #pragma HLS STREAM variable=layer9_out depth=1024
    #pragma HLS STABLE variable=layer9_out
    nnet::relu<layer6_t, layer9_t, relu_config9>(layer6_out, layer9_out); // activation_1

    hls::stream<layer43_t> layer43_out("layer43_out");
    #pragma HLS STREAM variable=layer43_out depth=1156
    #pragma HLS STABLE variable=layer43_out
    nnet::zeropad2d_cl<layer9_t, layer43_t, config43>(layer9_out, layer43_out); // zp2d_conv2d_2

    hls::stream<layer10_t> layer10_out("layer10_out");
    #pragma HLS STREAM variable=layer10_out depth=1024
    #pragma HLS STABLE variable=layer10_out
    nnet::conv_2d_cl<layer43_t, layer10_t, config10>(layer43_out, layer10_out, w10, b10); // conv2d_2
```

It seems to me that we have `conv2d_1  -[layer6_out]-> activation_1 -[layer9_out]-> zp2d_conv2d_2 -[layer43_out]-> conv2d_2`. Can you please point to where did you see the problem? It could be under my nose and I did not notice it.

Thank you,
Giuseppe

vloncar, 2021-03-22:
Hi Giuseppe,

The issue I see is just below the code portion you mention.

```c++
    hls::stream<layer51_t> layer51_cpy1("layer51_cpy1");
    #pragma HLS STREAM variable=layer51_cpy1 depth=1024
    hls::stream<layer51_t> layer51_cpy2("layer51_cpy2");
    #pragma HLS STREAM variable=layer51_cpy2 depth=1024
    nnet::clone_stream<layer14_t, layer51_t, 16384>(layer14_out, layer51_cpy1, layer51_cpy2); // clone_activation_2

    hls::stream<layer46_t> layer46_out("layer46_out");
    #pragma HLS STREAM variable=layer46_out depth=1089
    nnet::zeropad2d_cl<layer51_t, layer46_t, config46>(layer51_cpy1, layer46_out); // zp2d_conv2d_5

    hls::stream<layer44_t> layer44_out("layer44_out");
    #pragma HLS STREAM variable=layer44_out depth=1089
    nnet::zeropad2d_cl<layer46_t, layer44_t, config44>(layer46_out, layer44_out); // zp2d_conv2d_3

    hls::stream<layer15_t> layer15_out("layer15_out");
    #pragma HLS STREAM variable=layer15_out depth=121
    nnet::conv_2d_cl<layer44_t, layer15_t, config15>(layer44_out, layer15_out, w15, b15); // conv2d_3
```
You can see the two `zeropad2d_cl` functions in a row. As a consequence of an error in wiring `zp2d_conv2d_3`, `layer44_out` gets to be extra padded, while `layer21_out` is not padded. So at the end we have extra contents in `layer44_out` while at one point we try to read from the empty `layer21_out`.

GiuseppeDiGuglielmo, 2021-03-22:
I see. This may require some manual editing of the generated code I guess, although that may be error-prone. We have to think about it.

Thank you for noticing it,
Giuseppe

vloncar, 2021-03-22:
I have a potential fix. We need to add a proper check for the correct node [here](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/model/hls_model.py#L300), instead of taking the first one with `next()`. But I need to test if anything breaks by this. Should be ready tomorrow.

closed 2022-06-19 by jmduarte