fastmachinelearning/hls4ml #308: Predict function throws segmentation fault

ekellim, 2021-03-23:
Hello, 

I'm trying to convert a keras model to hls to deploy it on FPGA. 

After compilation of the hls model I use the predict function of hls4ml to compare the accuracy of the hls model with the original keras model. However when I use this function for my own model(s), it throws a Segmentation fault (core dumped) error.

Here is my model with configuration and output:

Interpreting Sequential
Topology:
Layer name: conv1d_input, layer type: Input
Layer name: conv1d, layer type: Conv1D
  -> Activation (relu), layer name: conv1d
Layer name: max_pooling1d, layer type: MaxPooling1D
Layer name: conv1d_1, layer type: Conv1D
  -> Activation (relu), layer name: conv1d_1
Layer name: max_pooling1d_1, layer type: MaxPooling1D
Layer name: conv1d_2, layer type: Conv1D
  -> Activation (relu), layer name: conv1d_2
Layer name: dense, layer type: Dense
  -> Activation (relu), layer name: dense
Layer name: dense_1, layer type: Dense
  -> Activation (softmax), layer name: dense_1
-----------------------------------
Configuration
Model
  Precision:         ap_fixed<16,6>
  ReuseFactor:       1
  Strategy:          Latency
LayerName
  conv1d_input
    Precision
      result:        ap_fixed<16,6>
  conv1d
    Precision
      weight:        ap_fixed<16,6>
      bias:          ap_fixed<16,6>
      result:        ap_fixed<16,6>
    ReuseFactor:     1
  conv1d_relu
    Precision:       ap_fixed<16,6>
    ReuseFactor:     1
    table_size:      1024
    table_t:         ap_fixed<18,8>
  max_pooling1d
    Precision:       ap_fixed<16,6>
  conv1d_1
    Precision
      weight:        ap_fixed<16,6>
      bias:          ap_fixed<16,6>
      result:        ap_fixed<16,6>
    ReuseFactor:     1
  conv1d_1_relu
    Precision:       ap_fixed<16,6>
    ReuseFactor:     1
    table_size:      1024
    table_t:         ap_fixed<18,8>
  max_pooling1d_1
    Precision:       ap_fixed<16,6>
  conv1d_2
    Precision
      weight:        ap_fixed<16,6>
      bias:          ap_fixed<16,6>
      result:        ap_fixed<16,6>
    ReuseFactor:     1
  conv1d_2_relu
    Precision:       ap_fixed<16,6>
    ReuseFactor:     1
    table_size:      1024
    table_t:         ap_fixed<18,8>
  dense
    Precision
      weight:        ap_fixed<16,6>
      bias:          ap_fixed<16,6>
      result:        ap_fixed<16,6>
    ReuseFactor:     1
  dense_relu
    Precision:       ap_fixed<16,6>
    ReuseFactor:     1
    table_size:      1024
    table_t:         ap_fixed<18,8>
  dense_1
    Precision
      weight:        ap_fixed<16,6>
      bias:          ap_fixed<16,6>
      result:        ap_fixed<16,6>
    ReuseFactor:     1
  dense_1_softmax
    Precision:       ap_fixed<16,6>
    ReuseFactor:     1
    table_size:      1024
    table_t:         ap_fixed<18,8>
-----------------------------------
Interpreting Sequential
Topology:
Layer name: conv1d_input, layer type: InputLayer, current shape: [[None, 193, 1]]
Layer name: conv1d, layer type: Conv1D, current shape: [[None, 193, 1]]
Layer name: max_pooling1d, layer type: MaxPooling1D, current shape: [[None, 187, 56]]
Layer name: conv1d_1, layer type: Conv1D, current shape: [[None, 62, 56]]
Layer name: max_pooling1d_1, layer type: MaxPooling1D, current shape: [[None, 57, 72]]
Layer name: conv1d_2, layer type: Conv1D, current shape: [[None, 28, 72]]
Layer name: dense, layer type: Dense, current shape: [[None, 26, 128]]
Layer name: dense_1, layer type: Dense, current shape: [[None, 128]]
Creating HLS model
Writing HLS project
Done
Segmentation fault (core dumped)

Is this a RAM issue? Anyone that can help?

Thanks!

note: I ran the code first on a 4GB RAM machine, afterwards on a 12GB RAM machine, got the error on both

jmitrevs, 2021-03-23:
Can you provide a bit more information on the steps you do?

ekellim, 2021-03-23:
**First I load the model and validation data:**

model = load_model('models/model_norm.h5')
X_test = np.load('models/X_test_norm.npy')
y_test = np.load('models/y_test_norm.npy')

**Then I generate a config and convert to a hls model in this way:**

config = hls4ml.utils.config_from_keras_model(model, granularity='name')
hls_model = hls4ml.converters.convert_from_keras_model(model,
                                                       hls_config=config,
                                                       output_dir='models/hls4ml_prj',
                                                       fpga_part='xc7z020clg400-1')

**Finally I compile the hls model, convert the X_test array to a contiguous array and run the predict function**

hls_model.compile()
X_test = np.ascontiguousarray(X_test)
y_hls = hls_model.predict(X_test) <-- This line throws the error

**These are the model and data I use:**
[model_norm.zip](https://github.com/fastmachinelearning/hls4ml/files/6191522/model_norm.zip)

vloncar, 2021-03-23:
You need to use `io_stream` for CNN models. You can pass it to [`convert_from_keras_model`](https://fastmachinelearning.org/hls4ml/autodoc/hls4ml.converters.html#hls4ml.converters.convert_from_keras_model)

ekellim, 2021-03-23:
> You need to use `io_stream` for CNN models. You can pass it to [`convert_from_keras_model`](https://fastmachinelearning.org/hls4ml/autodoc/hls4ml.converters.html#hls4ml.converters.convert_from_keras_model)

Thanks for the fast reply. Unfortunately I get the same error after applying this.

vloncar, 2021-03-23:
The issue is that we don't currently support applying `Dense` layer on a multidimensional input, such as `Dense` following `Conv2D` without applying `Flatten` layer first. The other thing is that this model is too large for hls4ml. It has over 100k parameters. I would suggest you start with something that is 10x smaller and build up.

ekellim, 2021-04-05:
Using a smaller model worked for me, is there a fixed max limit of parameters a model can have or does this depend on the kind of network?

closed 2021-04-27 by ekellim