fastmachinelearning/hls4ml #313: Issue with synthesizing example project

ThGkasios, 2021-04-02:
Hello!

I'm basically trying to figure out hls4ml, and I'm encountering some errors in the rtl synthesis step.

Namely:

```
***** C/RTL SYNTHESIS *****
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ...
ERROR: [HLS 200-70] Compilation errors found: In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:19:
In file included from D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\iostream:39:
In file included from D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ostream:39:
In file included from D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ios:39:
In file included from D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception:151:
D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception_ptr.h:132:13: error: unknown type name 'type_info'
      const type_info*
            ^
In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:19:
In file included from D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\iostream:39:
In file included from D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ostream:39:
In file included from D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ios:39:
In file included from D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception:151:
D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:62:5: error: the parameter for this explicitly-defaulted copy constructor is const, but a member or base requires it to be non-const
    nested_exception(const nested_exception&) = default;
    ^
D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:64:23: error: the parameter for this explicitly-defaulted copy assignment operator is const, but a member or base requires it to be non-const
    nested_exception& operator=(const nested_exception&) = default;
                      ^
D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:77:28: error: exception specification in declaration does not match previous declaration
  inline nested_exception::~nested_exception() = default;
                           ^
D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:66:20: note: previous declaration is here
    inline virtual ~nested_exception();
                   ^
D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:122:61: error: redefinition of default argument
    __throw_with_nested(_Ex&& __ex, const nested_exception* = 0)
                                                            ^ ~
D:/Tools/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:110:56: note: previous definition is here
    __throw_with_nested(_Ex&&, const nested_exception* = 0)
                                                       ^ ~
In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:21:
In file included from firmware/myproject.h:23:
In file included from D:/Tools/Xilinx/Vivado/2019.2/common/technology/autopilot\ap_int.h:367:
In file included from D:/Tools/Xilinx/Vivado/2019.2/common/technology/autopilot\ap_fixed.h:55:
D:/Tools/Xilinx/Vivado/2019.2/common/technology/autopilot\ap_fixed_base.h:838:13: warning: shift count is negative [-Wshift-count-negative]
      ret.V <<= (_AP_I - _AP_W);
            ^   ~~~~~~~~~~~~~~~
D:/Tools/Xilinx/Vivado/2019.2/common/technology/autopilot\ap_fixed_base.h:869:69: note: in instantiation of member function 'ap_fixed_base<50, 40, true, 5, 3, 0>::to_ap_int_base' requested here
  inline __attribute__((always_inline)) int to_int() const { return to_ap_int_base().to_int(); }
                                                                    ^
D:/Tools/Xilinx/Vivado/2019.2/common/technology/autopilot\ap_fixed_base.h:1042:71: note: in instantiation of member function 'ap_fixed_base<50, 40, true, 5, 3, 0>::to_int' requested here
  inline __attribute__((always_inline)) operator int() const { return to_int(); }
                                                                      ^
firmware/nnet_utils/nnet_activation.h:389:17: note: in instantiation of member function 'ap_fixed_base<50, 40, true, 5, 3, 0>::operator int' requested here
   data_round = (data_cache[jj]-data_cache[ii])*CONFIG_T::table_size/16;
                ^
firmware/nnet_utils/nnet_activation.h:421:9: note: in instantiation of function template specialization 'nnet::softmax_legacy<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, softmax_config9>' requested here
        softmax_legacy<data_T, res_T, CONFIG_T>(data, res);
        ^
firmware/myproject.cpp:90:5: note: in instantiation of function template specialization 'nnet::softmax<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, softmax_config9>' requested here
    nnet::softmax<layer8_t, result_t, softmax_config9>(layer8_out, layer9_out);
    ^
In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:21:
In file included from firmware/myproject.h:23:
In file included from D:/Tools/Xilinx/Vivado/2019.2/common/technology/autopilot\ap_int.h:367:
In file included from D:/Tools/Xilinx/Vivado/2019.2/common/technology/autopilot\ap_fixed.h:55:
D:/Tools/Xilinx/Vivado/2019.2/common/technology/autopilot\ap_fixed_base.h:838:13: warning: shift count is negative [-Wshift-count-negative]
      ret.V <<= (_AP_I - _AP_W);
            ^   ~~~~~~~~~~~~~~~
D:/Tools/Xilinx/Vivado/2019.2/common/technology/autopilot\ap_fixed_base.h:869:69: note: in instantiation of member function 'ap_fixed_base<51, 41, true, 5, 3, 0>::to_ap_int_base' requested here
  inline __attribute__((always_inline)) int to_int() const { return to_ap_int_base().to_int(); }
                                                                    ^
D:/Tools/Xilinx/Vivado/2019.2/common/technology/autopilot\ap_fixed_base.h:1042:71: note: in instantiation of member function 'ap_fixed_base<51, 41, true, 5, 3, 0>::to_int' requested here
  inline __attribute__((always_inline)) operator int() const { return to_int(); }
                                                                      ^
firmware/nnet_utils/nnet_activation.h:401:27: note: in instantiation of member function 'ap_fixed_base<51, 41, true, 5, 3, 0>::operator int' requested here
      int exp_res_index = exp_res[ii]*CONFIG_T::table_size/64;
                          ^
firmware/nnet_utils/nnet_activation.h:421:9: note: in instantiation of function template specialization 'nnet::softmax_legacy<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, softmax_config9>' requested here
        softmax_legacy<data_T, res_T, CONFIG_T>(data, res);
        ^
firmware/myproject.cpp:90:5: note: in instantiation of function template specialization 'nnet::softmax<ap_fixed<16, 6, 5, 3, 0>, ap_fixed<16, 6, 5, 3, 0>, softmax_config9>' requested here
    nnet::softmax<layer8_t, result_t, softmax_config9>(layer8_out, layer9_out);
    ^
2 warnings and 5 errors generated.
Failed during preprocessing.
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Fri Apr  2 16:14:58 2021...
```

I am using Windows 10, and my Vivado HLS version is 2019.2, but I've had this issue before.

Anyway this can be resolved?

vloncar, 2021-04-02:
This is a known issue on Windows and we have no solution for it. You should use Linux.

ThGkasios, 2021-04-02:
Is this something that can be solved by uninstalling and reinstalling the programms? I would prefer to do this if it can do anything.

vloncar, 2021-04-02:
No. Install Linux into a virtual machine and try Vivado there. You can generate projects with hls4ml in windows then run them in vivado in Linux

ThGkasios, 2021-04-02:
Thanks for the help!

closed 2021-04-02 by ThGkasios