fastmachinelearning/hls4ml #315: Getting Started example errors during execution

tianrui-wei, 2021-04-06:
I was trying to run the getting started example in this repository, and here is the error backtrace I got

```
Downloading example model files ...
{'OutputDir': 'my-hls-test', 'ProjectName': 'myproject', 'XilinxPart': 'xcku115-flvb2104-2-i', 'ClockPeriod': 5, 'Backend': 'Vivado', 'IOType': 'io_parallel', 'HLSConfig': {'Model': {'Precision': 'ap_fixed<16,6>', 'ReuseFactor': '1'}}, 'KerasJson': 'KERAS_3layer.json', 'KerasH5': 'KERAS_3layer_weights.h5'}
Interpreting Model
Topology:
Layer name: input_1, layer type: InputLayer, current shape: [[None, 16]]
Layer name: fc1_relu, layer type: Dense, current shape: [[None, 16]]
Layer name: fc2_relu, layer type: Dense, current shape: [[None, 64]]
Layer name: fc3_relu, layer type: Dense, current shape: [[None, 32]]
Layer name: output_softmax, layer type: Dense, current shape: [[None, 32]]
Creating HLS model
{   'keras': [   'KERAS_3layer_binarydense_relu_max.json',
                 'keras_bnn.json',
                 'jetTagger_Conv2D_Small.json',
                 'KERAS_conv2d_model.json',
                 'KERAS_dense_16x100x100x100x100x100x5.json',
                 'qkeras_mnist_dense.json',
                 'KERAS_dense_16x200x200x200x200x200x5.json',
                 'KERAS_3layer_ternary_small.json',
                 'qkeras_3layer.json',
                 'KERAS_conv1d_small.json',
                 'jetTagger_Conv2D_Small_NoBatchNorm.json',
                 'KERAS_3layer_binary_smaller.json',
                 'KERAS_conv1d.json',
                 'garnet_1layer.json',
                 'KERAS_3layer.json',
                 'KERAS_1layer.json',
                 'KERAS_3layer_batch_norm.json',
                 'garnet_3layer.json',
                 'KERAS_dense_16x500x500x500x500x500x5.json'],
    'onnx': [   'three_layer_bn_keras.onnx',
                'two_layer_pytorch.onnx',
                'conv1d_small_keras.onnx',
                'three_layer_bn_pytorch.onnx',
                'three_layer_keras.onnx',
                'dense_big_keras.onnx',
                'two_layer_keras.onnx',
                'conv2d_small_mp_keras.onnx',
                'conv2d_small_keras.onnx',
                'three_layer_pytorch.onnx'],
    'pytorch': ['two_layer_model.pt', 'three_layer_model.pt'],
    'tensorflow': ['3layer.pb']}
Exception ignored in: <bound method KerasFileReader.__del__ of <hls4ml.converters.keras_to_hls.KerasFileReader object at 0x7f58bd66c470>>
Traceback (most recent call last):
  File "/home/test/.local/lib/python3.6/site-packages/hls4ml/converters/keras_to_hls.py", line 18, in __del__
  File "/home/test/.local/lib/python3.6/site-packages/h5py/_hl/files.py", line 446, in close
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 275, in h5py.h5f.get_obj_ids
  File "h5py/h5i.pyx", line 40, in h5py.h5i.wrap_identifier
ImportError: sys.meta_path is None, Python is likely shutting down
```

I was wondering if anyone would be able to take a look at this. Thanks!

Duchstf, 2021-05-13:
What specific model were you trying to convert? And did you encounter the error when running this line?

```
hls_model = hls4ml.converters.keras_to_hls(config)
```

tianrui-wei, 2021-05-13:
I was using the model from the example
```python3
import hls4ml

#Fetch a keras model from our example repository
#This will download our example model to your working directory and return an example configuration file
config = hls4ml.utils.fetch_example_model('KERAS_3layer.json')

print(config) #You can print the configuration to see some default parameters

#Convert it to a hls project
hls_model = hls4ml.converters.keras_to_hls(config)

# Print full list of example models if you want to explore more
hls4ml.utils.fetch_example_list()
```

and the errors indeed happened during that line

closed 2026-10-25 by github-actions[bot]