fastmachinelearning/hls4ml #322: errors when using conv2d layers

Vincent-Zh, 2021-04-16:
I change your network to a simple ConvNet:

_model = Sequential()
model.add(Conv2D(8, 3, input_shape=(16,16,1), name='conv1', padding='same',
                 kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001)))
model.add(Activation(activation='relu', name='relu1'))
model.add(Conv2D(8, 3, name='conv2', padding='same',
                 kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001)))
model.add(Activation(activation='relu', name='relu2'))
model.add(Conv2D(1, 3, name='conv3', padding='same', 
                 kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001)))_

When I execute the final command: 

_hls_model.build(csim=True, synth=True, cosim=True, validation=False, export=True)_

An error happened as follows:

ERROR: [XFORM 203-504] Stop unrolling loop 'ConvOutHeight' (firmware/nnet_utils/nnet_conv2d.h:245) in function 'nnet::conv_2d_latency_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config8>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

I don't know why this error happened as I only changed the network. BTW, I'm not familiar with vivado and fpga.

thesps, 2021-04-16:
I transferred this issue from the hls4ml-tutorial repo since it's more relevant here.

vloncar, 2021-04-16:
Set `IOType` to `io_stream`, for example `hls4ml.converters.convert_from_keras_model(model, ..., io_type='io_stream')`

Vincent-Zh, 2021-04-19:
TypeError: convert_from_keras_model() got an unexpected keyword argument 'io_type'

I check the package source files, _converters.__init__.py_, 
I found this function does not have an argument -- 'io_type':

_def convert_from_keras_model(model, output_dir='my-hls-test', project_name='myproject',
    fpga_part='xcku115-flvb2104-2-i', clock_period=5, hls_config={}):_

Vincent-Zh, 2021-04-19:
I know what happened. In the _environment.yml_ of hls4ml-tutorial, the version of hls4ml is 0.4.0. 
But the lastest version should be 0.5.0. I update this package then it works.
Thanks, @vloncar

mackncheesiest, 2021-06-13:
With `hls4ml==0.5.0`, I'm seeing this same issue when using the command line interface + [KERAS_conv2d](https://github.com/fastmachinelearning/example-models/blob/master/keras/KERAS_conv2d_model.json) sample model. It looks to be because of the same `IOType` reason as above.

Essentially,
```
hls4ml config -m KERAS_conv2d_model.json -w KERAS_conv2d_model_weights.h5 -o KERAS_conv2d_configured.yml
```
produces a YAML file with `IOType: io_parallel` instead of `IOType: io_stream`, so the same error happens when I then proceed to run `hls4ml convert -> hls4ml build`

Making that change in the generated YAML file makes the error stop happening

behnamarefy, 2023-10-28:
i have the same problem im using vivado 2019.2 and my hls4ml version is 0.7.1 and my model is like :                                                   
Interpreting Model
Topology:
Layer name: input_2, layer type: InputLayer, input shapes: [[None, 2, 128, 1]], output shape: [None, 2, 128, 1]
Layer name: conv_0, layer type: Conv2D, input shapes: [[None, 2, 128, 1]], output shape: [None, 2, 128, 256]
Layer name: bn_conv_0, layer type: BatchNormalization, input shapes: [[None, 2, 128, 256]], output shape: [None, 2, 128, 256]
Layer name: conv_act_0, layer type: Activation, input shapes: [[None, 2, 128, 256]], output shape: [None, 2, 128, 256]
Layer name: pool_0, layer type: MaxPooling2D, input shapes: [[None, 2, 128, 256]], output shape: [None, 2, 64, 256]
Layer name: conv_1, layer type: Conv2D, input shapes: [[None, 2, 64, 256]], output shape: [None, 2, 64, 128]
Layer name: bn_conv_1, layer type: BatchNormalization, input shapes: [[None, 2, 64, 128]], output shape: [None, 2, 64, 128]
Layer name: conv_act_1, layer type: Activation, input shapes: [[None, 2, 64, 128]], output shape: [None, 2, 64, 128]
Layer name: pool_1, layer type: MaxPooling2D, input shapes: [[None, 2, 64, 128]], output shape: [None, 2, 32, 128]
Layer name: conv_2, layer type: Conv2D, input shapes: [[None, 2, 32, 128]], output shape: [None, 2, 32, 64]
Layer name: bn_conv_2, layer type: BatchNormalization, input shapes: [[None, 2, 32, 64]], output shape: [None, 2, 32, 64]
Layer name: conv_act_2, layer type: Activation, input shapes: [[None, 2, 32, 64]], output shape: [None, 2, 32, 64]
Layer name: pool_2, layer type: MaxPooling2D, input shapes: [[None, 2, 32, 64]], output shape: [None, 2, 16, 64]
Layer name: conv_3, layer type: Conv2D, input shapes: [[None, 2, 16, 64]], output shape: [None, 2, 16, 64]
Layer name: bn_conv_3, layer type: BatchNormalization, input shapes: [[None, 2, 16, 64]], output shape: [None, 2, 16, 64]
Layer name: conv_act_3, layer type: Activation, input shapes: [[None, 2, 16, 64]], output shape: [None, 2, 16, 64]
Layer name: pool_3, layer type: MaxPooling2D, input shapes: [[None, 2, 16, 64]], output shape: [None, 2, 8, 64]
Layer name: flatten_1, layer type: Reshape, input shapes: [[None, 2, 8, 64]], output shape: [None, 1024]
Layer name: dense_0, layer type: Dense, input shapes: [[None, 1024]], output shape: [None, 128]
Layer name: bn_dense_0, layer type: BatchNormalization, input shapes: [[None, 128]], output shape: [None, 128]
Layer name: dense_act_0, layer type: Activation, input shapes: [[None, 128]], output shape: [None, 128]
Layer name: dense_1, layer type: Dense, input shapes: [[None, 128]], output shape: [None, 64]
Layer name: bn_dense_1, layer type: BatchNormalization, input shapes: [[None, 64]], output shape: [None, 64]
Layer name: dense_act_1, layer type: Activation, input shapes: [[None, 64]], output shape: [None, 64]
Layer name: output_dense, layer type: Dense, input shapes: [[None, 64]], output shape: [None, 11]
Layer name: output_softmax, layer type: Softmax, input shapes: [[None, 11]], output shape: [None, 11]
and i got the follwing error when i ran hls_model.build(csim= False  , synth=True , vsynth = F
alse): ERROR: [XFORM 203-504] Stop unrolling loop 'MultLoop' (firmware/nnet_utils/nnet_dense_resource.h:52) in function 'nnet::conv_2d_cl<nnet::array<ap_fixed<22, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 64u>, nnet::array<ap_fixed<22, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 64u>, config17>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "
 @vloncar vloncar @thesps  @Nathan-zh

layson-inventor, 2023-12-27:
I also encountered the same problem, I have tried the above specified io_type, it dosen't work.

closed 2026-10-26 by github-actions[bot]