fastmachinelearning/hls4ml #326: qkeras conversion to hls4ml accuracy degradation

ekellim, 2021-04-27:
Hello, 

First of all thanks for the nice work and updates. Recently I tried to convert a CNN trained in qkeras to a hls4ml model, but the resulting accuracy was very poor (random outputs). I was hoping that the updates in #321 would help me solve this issue but `set_data_types_from_keras_model()` doesn't improve the accuracy, while `set_accum_from_keras_model() `throws following error: 

```
File "test.py", line 114, in <module>
    hls4ml.utils.config.set_accum_from_keras_model(config, model)
  File "/usr/local/lib/python3.6/dist-packages/hls4ml-0.5.1-py3.6.egg/hls4ml/utils/config.py", line 150, in set_accum_from_keras_model
    type_i = VivadoBackend.convert_precision_string(None, previous_layer_config['Precision']['result'])
KeyError: 'result'
```

Any recommendations on how to solve this?

Some more information:

**Model summary:** 

```
Model: "sequential"
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
conv1d_1 (QConv1D)           (None, 187, 28)           224       
_________________________________________________________________
activation1 (QActivation)    (None, 187, 28)           0         
_________________________________________________________________
max_pooling1d (MaxPooling1D) (None, 62, 28)            0         
_________________________________________________________________
conv1d_2 (QConv1D)           (None, 57, 36)            6084      
_________________________________________________________________
activation2 (QActivation)    (None, 57, 36)            0         
_________________________________________________________________
max_pooling1d_1 (MaxPooling1 (None, 28, 36)            0         
_________________________________________________________________
conv1d_3 (QConv1D)           (None, 26, 48)            5232      
_________________________________________________________________
activation3 (QActivation)    (None, 26, 48)            0         
_________________________________________________________________
flatten (Flatten)            (None, 1248)              0         
_________________________________________________________________
dense_1 (QDense)             (None, 20)                24980     
_________________________________________________________________
activation4 (QActivation)    (None, 20)                0         
_________________________________________________________________
dropout (Dropout)            (None, 20)                0         
_________________________________________________________________
dense_2 (QDense)             (None, 10)                210       
_________________________________________________________________
softmax (Activation)         (None, 10)                0         

```
hls4ml branch: https://github.com/maksgraczyk/hls4ml/tree/auto-type-setup

Code to reproduce the issue in attachment: [test_qkeras.zip](https://github.com/fastmachinelearning/hls4ml/files/6387647/test_qkeras.zip)

maksgraczyk, 2021-04-28:
Hello, thank you for raising the issue. I currently have a similar problem as yours, so this is what I'm working on at the moment. The `KeyError` in `set_accum_from_keras_model()` should be gone now, could you try running it again?

ekellim, 2021-04-28:
Hello, yes the problem is solved now! When using `set_accum_from_keras_model()`, the error is gone and the accuracy is the same as the qkeras model. Thanks a lot!

closed 2021-04-28 by maksgraczyk