fastmachinelearning/hls4ml #335: Bad accuracy when using 'resource' strategy in combination with qkeras and set_accum_from_keras_model()

ekellim, 2021-05-13:
Hello, 

When I use the `set_accum_from_keras_model()` function from #321 in combination with `config['Model']['Strategy'] = 'resource'`, the resulting accuracy is very bad. With the 'latency' strategy everything works fine however. Is this normal behaviour?

hls4ml branch: [https://github.com/maksgraczyk/hls4ml/tree/auto-type-setup](url)
code in attachment: [test_qkerasv2.zip](https://github.com/fastmachinelearning/hls4ml/files/6472758/test_qkerasv2.zip)

ekellim, 2021-05-17:
I found out it had nothing to do with `set_accum_from_keras_model()`. The problem persists without this function when converting from qkeras to hls4ml and when specifying `resource` as strategy. I noticed a similar issue where the uncorrect transposing of the weights for the resource strategy was fixed, I tested this on a normal (keras) model and this works indeed. So I think the problem is somewhere in converting a qkeras model with the `resource` strategy.

thesps, 2021-05-18:
I was able to see something similar, specifically in a context where my kernel quantizer had `alpha='auto'`. It seems like the `FactorizeAlpha` optimizer pass wasn't handling `resource` strategy properly. (Setting `alpha=1` for the model I got the correct model accuracy).
I prepared a fix, could you try out [this branch](https://github.com/thesps/hls4ml/tree/resource-strat-alpha) and let me know if it fixes the issue for you?

ekellim, 2021-05-18:
I tried with the new branch but it didn't fix the accuracy unfortunately. Also setting all `alpha=1` does seem to improve the accuracy but it is not the same yet as with the `Latency` strategy.

vloncar, 2021-05-18:
I think the fix in that branch is not correct for Conv1D/2D, it only works for dense. Problem is `weights.T` is not the correct transpose. 
For example, a 4D weight tensor as in Conv2D:
```
import numpy as np
w = np.arange(1,121).reshape(2,3,4,5)
print('What we get:', w.T.shape)
print('What we need:', np.transpose(w, axes=[3,0,1,2]).shape)
```
Prints out:
```
What we get: (5, 4, 3, 2)
What we need: (5, 2, 3, 4)

thesps, 2021-05-19:
That makes sense. I'm wondering whether to try to implement this now - checking for resource strategy and layer type in the pass - or wait for when the resource strategy transpose op is moved to the backend. There are other passes that might be affected by this, like [here](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/model/optimizer/passes/bn_fuse.py#L24).

We could also potentially simplify this in the short term on current state of master branch by moving the weight transpose op out of the layer initialisation, into an optimizer pass that goes last in the list.

vloncar, 2021-05-19:
I don't think we should wait. These changes are easy to pick up in the new branch (I just need to remove that check altogether). In general, the permutation is to put the last dim first and keep the rest, then we can pass that to `np.transpose`. For example:

```
ndim = len(weights.shape) - 1
perm = [ndim] + [i for i in range(ndim)]
new_weights = np.transpose(weights, axes=perm)
```
This should work for all layers. Can you add to your branch and make a PR?

thesps, 2021-05-20:
I've made the change, plus another that was uncovered by @maksgraczyk on #328 regarding the inferred shape of the scale tensor for a CNN when it is specified as a scalar. 

@ekellim could you try out [the branch](https://github.com/thesps/hls4ml/tree/resource-strat-alpha)?

ekellim, 2021-05-20:
It works fine now. Thanks for the quick update, keep up the great work.

closed 2021-05-20 by ekellim