fastmachinelearning/hls4ml #351: Bug with #PRAGMA HLS STABLE

nicologhielmetti, 2021-07-01:
During the development of `VivadoAccelerator` [backend](https://github.com/fastmachinelearning/hls4ml/pull/349) I noticed that the `#PRAGMA HLS STABLE` causes unexpected outputs for the models during cosim. I created 4 tests which are the combinations of these settings:
`{io_stream, io_parallel} x {stable_pragma, no_stable_pragma}`. You can find the generated projects [here](https://github.com/nicologhielmetti/hls4ml/tree/stable_pragma_tests/test-vivado-accelerator).
You can see, comparing the files generated by csim and cosim, that the validation doesn't result as passed for all the cases that contain `#PRAGMA HLS STABLE`.
In particular:
_`{stable_pragma, io_stream}` shows duplicates rows in the cosim output_
**cosim**
```
 { data: 0.366211, last: 0 }
 { data: 0.104492, last: 0 }
 { data: 0.0869141, last: 0 }
 { data: 0.267578, last: 0 }
 { data: 0.196289, last: 1 }
 
 { data: 0.366211, last: 0 }
 { data: 0.104492, last: 0 }
 { data: 0.0869141, last: 0 }
 { data: 0.267578, last: 0 }
 { data: 0.196289, last: 1 }
``` 
**csim**
```

 { data: 0.130859, last: 0 }
 { data: 0.130859, last: 0 }
 { data: 0.428711, last: 0 }
 { data: 0.167969, last: 0 }
 { data: 0.167969, last: 1 }
 
 { data: 0.366211, last: 0 }
 { data: 0.104492, last: 0 }
 { data: 0.0869141, last: 0 }
 { data: 0.267578, last: 0 }
 { data: 0.196289, last: 1 }
```
_`{stable_pragma, io_parallel}` shows slighlty different values in the cosim output_
**cosim**
```
 { data: 0.136719, last: 0 }
 { data: 0.136719, last: 0 }
 { data: 0.451172, last: 0 }
 { data: 0.175781, last: 0 }
 { data: 0.113281, last: 1 }
 
 { data: 0.347656, last: 0 }
 { data: 0.0996094, last: 0 }
 { data: 0.0820312, last: 0 }
 { data: 0.253906, last: 0 }
 { data: 0.238281, last: 1 }
```
**csim**
```
 { data: 0.130859, last: 0 }
 { data: 0.130859, last: 0 }
 { data: 0.428711, last: 0 }
 { data: 0.167969, last: 0 }
 { data: 0.167969, last: 1 }
 
 { data: 0.366211, last: 0 }
 { data: 0.104492, last: 0 }
 { data: 0.0869141, last: 0 }
 { data: 0.267578, last: 0 }
 { data: 0.196289, last: 1 }
```

thesps, 2021-07-01:
I'd just add that we saw that the appearance or not of this issue in the output seems to depend on the ReuseFactor, which is why in other cases it seems like there is no issue.

jmduarte, 2021-07-01:
@GiuseppeDiGuglielmo saw the same issue earlier, which is why we removed these `#pragma` statements from our official tiny MLPerf submission.

I could see it depending on the reuse factor... For the tiny MLPerf submission we were always using the max reuse factor possible for each layer.

ynozik, 2021-07-12:
I mentioned this issue in a separate thread (about a pull request for a CNN tutorial notebook) but I have encountered an issue where it appears impossible to use the `Resource` strategy under any circumstances (regardless of `ResourceFactor`, the size of the network, etc) because of the following error message when attempting to build:

---------------------------------------------------------------------------

***** C/RTL SYNTHESIS *****
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ... 
WARNING: [HLS 214-114] Since the only kind of statements allowed in a dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: firmware/nnet_utils/nnet_dense_latency.h:64:9
WARNING: [HLS 214-104] Only for-loops and functions support the dataflow: firmware/nnet_utils/nnet_dense_latency.h:76:9
WARNING: [HLS 214-114] Since the only kind of statements allowed in a dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: firmware/myproject.cpp:35:2
WARNING: [HLS 200-471] Dataflow form checks found 3 issue(s) in file firmware/myproject.cpp
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer2_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer5_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer6_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer7_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer10_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer11_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer12_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer15_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer16_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer17_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer20_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer21_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer24_out' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS STABLE variable=&layer25_out' is not a valid pragma.
Pragma processor failed: 
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

---------------------------------------------------------------------------

I'm not sure if this is related or maybe a separate issue with defining the network configuration.

vloncar, 2021-07-12:
That pragma is introduced in Vivado 2019.1, your version is too old. We support versions 2019.1-2020.1.

ynozik, 2021-07-12:
I see, thanks for the quick response. Perhaps the documentation could be updated then? https://fastmachinelearning.org/hls4ml/status.html Here it states that versions since 2018.2 are supported, which is where I had checked to see that my version is supported.

vloncar, 2021-07-12:
I agree, it should be updated. Can you open an issue for that? Thanks.

closed 2021-09-10 by vloncar