fastmachinelearning/hls4ml #352: Sepconv1D needs updating

thesps, 2021-07-02:
I think the `nnet_sepconv1d_stream.h` needs an update since the recent CNN merge that introduces the encoded and line buffer implementations as configurable options.

[This function](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/templates/vivado/nnet_utils/nnet_sepconv1d_stream.h#L42) is no longer defined anywhere, nor [this one](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/templates/vivado/nnet_utils/nnet_sepconv1d_stream.h#L60).

I think the relevant functions are implemented, but now with "encoded" or "buffer" in the function name too.

thaarres, 2021-07-02:
I don't know if it is related to this but I get the following when trying to synthesize a 1x1 Conv2D:
```

***** C/RTL SYNTHESIS *****
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ... 
ERROR: [HLS 200-70] Compilation errors found: In file included from firmware/myproject.cpp:1:
firmware/myproject.cpp:68:2: error: no matching function for call to 'pointwise_conv_2d_cl'
 nnet::pointwise_conv_2d_cl<layer2_t, layer12_t, config12>(layer2_out, layer12_out, w12, b12);
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
firmware/nnet_utils/nnet_sepconv2d_stream.h:79:6: note: candidate function [with data_T = ap_fixed<16, 6, 5, 3, 0>, res_T = ap_fixed<16, 6, 5, 3, 0>, CONFIG_T = config12] not viable: no known conversion from 'layer2_t [24]' to 'hls::stream<ap_fixed<16, 6, 5, 3, 0> > &' for 1st argument; 
void pointwise_conv_2d_cl(
     ^
```

vloncar, 2021-07-02:
@thesps Thanks, I made a PR to fix it.

@thaarres you forgot the `io_stream`

thaarres, 2021-07-04:
Thanks a lot @vloncar! I confirm that the Conv1D works again on vloncar:cnn_sepconv_fix (why is that header actually being loaded for plain Conv1D?). About the io, I indeed wanted to try io_parallel to get around the problem with sepconv1d (tiny tiny model), didn't realise that's not possible. Thanks for the feedback!