fastmachinelearning/hls4ml #376: Fix keras_api test

jmduarte, 2021-08-12:
_Originally posted by @thesps in https://github.com/fastmachinelearning/hls4ml/issues/362#issuecomment-897581793_

Fix keras_api test:
`FAILED test_keras_api.py::test_activation_thresholdedrelu[ThresholdedReLU] - AssertionError:`
This one I'm not actually sure about, I'm commenting it out for now but again with a note to fix the issue and include the test.

spacenets7, 2026-11-02:
I’d like to investigate the remaining direct ThresholdedReLU test coverage. I noticed that the activation test still excludes it, while there is now separate coverage for ReLU(threshold=...). I’ll check whether the original failure still occurs, using deterministic inputs below, at, and above the threshold. Is anyone currently working on this?

spacenets7, 2026-11-02:
I completed the initial checks on 863a4e9. Deterministic ThresholdedReLU boundary tests passed for Vivado and Vitis, with both parallel and stream I/O.

Restoring the original activation test body produced numerical mismatches. However, a reduced Dense → ThresholdedReLU example matched an independent fixed-point oracle at both tested precisions. Quantization can move the Dense output across the discontinuous threshold and produce a large difference from floating-point Keras.

I haven’t established a kernel defect or explained every original mismatch. Would a deterministic, precision-aware regression test be a useful contribution here?