fastmachinelearning/hls4ml #387: YAML config file using only h5 file from Keras's model

IanCheng18, 2021-08-29:
I am trying to use ONLY the .h5 weight file from Keras's model.save().
I simply deleted the "KerasJson: keras/KERAS_3layer.json" entire line.

```
KerasH5: boats10_weights_final.h5
#InputData: keras/KERAS_3layer_input_features.dat
#OutputPredictions: keras/KERAS_3layer_predictions.dat
OutputDir: output
ProjectName: boat_project
XilinxPart: AES-ULTRA96-V2-G # xcku115-flvb2104-2-i
ClockPeriod: 5
Backend: Vivado

IOType: io_parallel # options io_serial/io_parallel
HLSConfig:
  Model:
	Precision: ap_fixed<16,6>
	ReuseFactor: 1
#  LayerType:
#	Dense:
#  	ReuseFactor: 2
#  	Strategy: Resource
#  	Compression: True
```

Then I am always getting the following error:

```
 File "/home/yich541b/Desktop/yolov3/python-venv/hls4ml/lib/python3.6/site-packages/hls4ml/converters/keras_to_hls.py", line 222, in keras_to_hls
	raise ValueError('No model found in config file.')
ValueError: No model found in config file.
```

If I am only providing the .h5 file, how should I deal with the first line in `keras-config.yml?`

jmitrevs, 2021-08-30:
I think you can feed in either the full model h5 or the weights + json, but just just the weights.

IanCheng18, 2021-08-30:
Really appreciate your fast support, but the issue wasn't solved. :(
I used Keras's **model.save()** function to save my model, is this considered as the full model h5? This is the only file that was outputted after training my model.

I used [qqwweee's code](https://github.com/qqwweee/keras-yolo3) with merely no modification, which only has one h5 file (trained_weights_final.h5) saved after training.

I tried to feed in the full model h5 file as following: (not sure if this is the correct way??)

```
KerasJson: boats10_weights_final.h5
KerasH5:   boats10_weights_final.h5
#InputData: keras/KERAS_3layer_input_features.dat
#OutputPredictions: keras/KERAS_3layer_predictions.dat
OutputDir: my-hls-test
ProjectName: myproject
XilinxPart: xcku115-flvb2104-2-i
ClockPeriod: 5
Backend: Vivado

IOType: io_parallel # options: io_serial/io_parallel
HLSConfig:
  Model:
    Precision: ap_fixed<16,6>
    ReuseFactor: 1
#  LayerType:
#    Dense:
#      ReuseFactor: 2
#      Strategy: Resource
#      Compression: True
```

Which the following error appeared: 

```
File "/home/yich541b/Desktop/yolov3/python-venv/hls4ml/lib/python3.6/site-packages/hls4ml/converters/keras_to_hls.py", line 214, in keras_to_hls
	model_arch = json.load(json_file)
  File "/usr/lib/python3.6/json/__init__.py", line 296, in load
	return loads(fp.read(),
  File "/usr/lib/python3.6/codecs.py", line 321, in decode
	(result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x89 in position 0: invalid start byte
```

IanCheng18, 2021-08-30:
> I think you can feed in either the full model h5 or the weights + json, but not just the weights. I know in Keras I have done either:
> 
> ```python
> model = keras.models.load_model(MODEL)
> ```

You are using Python API to build the model, correct? 
So based on your method, it should be as following (please correct me if needed, many thanks)?

```
import hls4ml
config = hls4ml.utils.config_from_keras_model(model = keras.models.load_model(MODEL), granularity='model')
```

I am probably asking a dumb question, but where should I mention the h5 file name??

jmitrevs, 2021-08-30:
This is the jupyter notebook I have:
```python
from tensorflow import keras

MODEL = "policy_model_e940/policy_model_e940_.modelall.h5"
model = keras.models.load_model(MODEL)

#WEIGHTS = "MLP_model/final_policy_model.weights.h5"
#DESC = "MLP_model/final_policy_model.json"

#WEIGHTS = "untrained_model/initialized_weights.h5"
#DESC = "untrained_model/untrained_model.json"

#with open(DESC) as fl:
#    desc = "\n".join(fl.readlines())
#model = keras.models.model_from_json(desc)
#model.load_weights(WEIGHTS)
```
with things currently not used being commented out. This is followed by:
```python
config = hls4ml.utils.config_from_keras_model(model, granularity='model')
```
However, it should work with yml file, too. Based on https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/converters/keras_to_hls.py#L202, see if maybe you should use KerasModel as the config name. I didn't try it or look too carefull at the code.

IanCheng18, 2021-08-30:
Just a quick question, by any chance, are you using [qqwweee's keras-yolov3](https://github.com/qqwweee/keras-yolo3) for trianing?

closed 2021-08-31 by IanCheng18