fastmachinelearning/hls4ml #389: Simulation and weights reloading

rishab63, 2021-09-01:
Hello, I used hls4ml to convert a simple time series based MLP into HLS which I then synthesized using Vivado HLS.  I now want to test/simulate this design. I went through some of the tutorials where they validate the hls code using hls_model.predict(). However, I cant seem to compile my hls model. I saw somewhere that it is not supported on Windows. If so, how can I validate my design, should I create a testbench for cosimulation or create an RTL testbench in verilog to validate the resulting verilog implementation?

I also wanted to know if there is a provision to reload weights after inference, because the output generated seems to take the weight and bias values into account while translating the network.

jmitrevs, 2021-09-01:
About the second question, the default is to have fixed weights. This way, for example if you have a sparse model, you can take advantage to optimize the model to reduce resources. There is some development towards having models with reloadable weights, but I don't believe there is anything in the main repository. It becomes a bit more of a system design issue--how do you want to update the weights?

rishab63, 2021-09-01:
@jmitrevs  Thanks for the reply. By reloading of weights, I meant if there is way where I could implement a generic network and then load the weights stored FPGA memory(BRAM).

jmitrevs, 2021-09-01:
I was referring to weights loaded from outside the FPGA without reprogramming the whole FPGA. There is no official support currently, but people are looking at it. The standard setup is that the weights are fixed in the model. Unless you have a reason to need to update weights at runtime, this is preferred, since more optimizations are possible.

vloncar, 2021-09-01:
@rishab63 If you put `BramFactor` into the config, every weight array with more elements than the specified threshold will be treated as external BRAM port. So if you put the value of `1` every weight will be an external BRAM port. You must use `io_stream` and `Resource` strategy for this to work. Note that the feature is _very_ experimental.

closed 2021-11-02 by rishab63

GiulioMa, 2021-12-09:
Hi @rishab63,

Have you managed to solve the issue? I'm facing the same problem.

rishab63, 2021-12-10:
Hey @GiulioMa 

I think it can be possible like mentioned in the comments above(using `BRAM factor` and setting a threshold of `1`, `io_stream ` and `Resource ` strategy).  But, I did not try this yet fearing it might negatively impact **latency** (not desired for my experiment). However, I might try this and see what's possible because reloading weights could be really key in my project. Did  you explore this? I am curious to see the results and possibilities.

GiulioMa, 2021-12-10:
Hi @rishab63 
No, I haven't tried it yet. 
But if you find an alternative that do not sacrifice latency, let me know! I'll do the same if I found something.