fastmachinelearning/hls4ml #390: example-models,  C/RTL co-simulation error

IanCheng18, 2021-09-01:
I am trying to run the example-models (in Linux Ubuntu 18.04 with command line)following the official website of [HLS4ML QuickStart](https://fastmachinelearning.org/hls4ml/setup.html#quick-start)

Everything runs fine until I run:
`hls4ml build -p my-hls-test -a`

I faced the following error, this is just part of the error code. Complete version is in the attachment.
Did not find out a solution yet.
Thanks for your help and suggestions in advance.

```
***** C/RTL SIMULATION *****
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [COSIM 212-47] Using XSIM for RTL simulation.
INFO: [COSIM 212-14] Instrumenting C test bench ...
   Build using "/data/Xilinx/Vivado/2018.3/tps/lnx64/gcc-6.2.0/bin/g++"
   Compiling myproject_test.cpp_pre.cpp.tb.cpp
cosim.tv.mk:64: recipe for target 'obj/myproject_test.cpp_pre.cpp.tb.o' failed
In file included from /data/Xilinx/Vivado/2018.3/include/floating_point_v7_0_bitacc_cmodel.h:148:0,
                 from /data/Xilinx/Vivado/2018.3/include/hls_fpo.h:186,
                 from /data/Xilinx/Vivado/2018.3/include/hls_half.h:44,
                 from /data/Xilinx/Vivado/2018.3/include/etc/ap_private.h:90,
                 from /data/Xilinx/Vivado/2018.3/include/ap_common.h:641,
                 from /data/Xilinx/Vivado/2018.3/include/ap_int.h:54,
                 from /home/yich541b/Desktop/yolov3/example-models/my-hls-test-20183/firmware/myproject.h:23,
                 from /home/yich541b/Desktop/yolov3/example-models/my-hls-test-20183/myproject_test.cpp:28:
/data/Xilinx/Vivado/2018.3/include/mpfr.h:181:9: error: \u2018__gmp_const\u2019 does not name a type
 typedef __gmp_const __mpfr_struct *mpfr_srcptr;
         ^~~~~~~~~~~
/data/Xilinx/Vivado/2018.3/include/mpfr.h:181:48: error: expected constructor, destructor, or type conversion before \u2018;\u2019 token
 typedef __gmp_const __mpfr_struct *mpfr_srcptr;
                                                ^
In file included from /data/Xilinx/Vivado/2018.3/include/floating_point_v7_0_bitacc_cmodel.h:148:0,
                 from /data/Xilinx/Vivado/2018.3/include/hls_fpo.h:186,
                 from /data/Xilinx/Vivado/2018.3/include/hls_half.h:44,
                 from /data/Xilinx/Vivado/2018.3/include/etc/ap_private.h:90,
                 from /data/Xilinx/Vivado/2018.3/include/ap_common.h:641,
                 from /data/Xilinx/Vivado/2018.3/include/ap_int.h:54,
                 from /home/yich541b/Desktop/yolov3/example-models/my-hls-test-20183/firmware/myproject.h:23,
                 from /home/yich541b/Desktop/yolov3/example-models/my-hls-test-20183/myproject_test.cpp:28:
/data/Xilinx/Vivado/2018.3/include/mpfr.h:244:2: error: \u2018__gmp_const\u2019 does not name a type
 __MPFR_DECLSPEC __gmp_const char * mpfr_get_version _MPFR_PROTO ((void));
  ^~~~~~~~~~~
/data/Xilinx/Vivado/2018.3/include/mpfr.h:245:2: error: \u2018__gmp_const\u2019 does not name a type
 __MPFR_DECLSPEC __gmp_const char * mpfr_get_patches _MPFR_PROTO ((void));
  ^~~~~~~~~~~
/data/Xilinx/Vivado/2018.3/include/mpfr.h:260:2: error: \u2018__gmp_const\u2019 does not name a type
 __MPFR_DECLSPEC __gmp_const char *
  ^~~~~~~~~~~
/data/Xilinx/Vivado/2018.3/include/mpfr.h:299:19: error: \u2018mpfr_srcptr\u2019 was not declared in this scope
   mpfr_can_round _MPFR_PROTO ((mpfr_srcptr, mpfr_exp_t, mpfr_rnd_t, mpfr_rnd_t,
                   ^~~~~~~~~~~
```
[2018.3_error.txt](https://github.com/fastmachinelearning/hls4ml/files/7091727/2018.3_error.txt)

jmitrevs, 2021-09-01:
Just to make sure, you ran the following steps:

```
% git clone https://github.com/fastmachinelearning/example-models.git
% cd example-models/
% hls4ml convert -c keras-config.yml
```
and then this command fails:
```
% hls4ml build -p my-hls-test -a
```
Is that correct? I tried it using the master version of hls4ml and Vivado 2019.2 on Scientific Linux (approx = CentOS) 7.9 and it passes the C/RTL simulation step without a problem. Do you have access to a newer version of Vivado?

IanCheng18, 2021-09-01:
Yes, I cloned the example-models folder, cd to folder, and did the conversion.
You can see it in this error code file. 
[2018.3_error.txt](https://github.com/fastmachinelearning/hls4ml/files/7092334/2018.3_error.txt)

Also tried on Vivado 2019.1 and same error occurred.

thesps, 2021-09-01:
The problem is related to the `gmp` library & an issue with Vivado HLS, see [here](http://svn.clifford.at/handicraft/2017/vivadobugs/vivadobug04.txt) for example.

closed 2026-10-24 by github-actions[bot]