fastmachinelearning/hls4ml #392: README example not working

bluepra, 2021-09-04:
Using hls4ml 0.5.0, python 3.8.11, tensorflow 2.3.1, and Vivado HLS 2020.1.

I tried to replicate the README's example word for word, but when I run the code, the process fails during the build command step. Here is the code I am running:

import hls4ml
config = hls4ml.utils.fetch_example_model('KERAS_3layer.json')
hls_model = hls4ml.converters.keras_to_hls(config)
hls_model.build()

The build command generates a directory called my-hls-test which consists of multiple files and folders, one being vivado_hls.log. This log file consists of the output of the build command. I have pasted the last few lines before the program runs into the error:

INFO: [HLS 200-489] Unrolling loop 'ResetAccum' (firmware/nnet_utils/nnet_dense_latency.h:101) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' completely with a factor of 64.
INFO: [HLS 200-489] Unrolling loop 'Accum1' (firmware/nnet_utils/nnet_dense_latency.h:109) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' completely with a factor of 16.
INFO: [HLS 200-489] Unrolling loop 'Accum2' (firmware/nnet_utils/nnet_dense_latency.h:113) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' completely with a factor of 64.
INFO: [HLS 200-489] Unrolling loop 'Result' (firmware/nnet_utils/nnet_dense_latency.h:120) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' completely with a factor of 64.
INFO: [XFORM 203-131] Reshaping array 'input_1.V' (firmware/myproject.cpp:25) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'layer9_out.V' (firmware/myproject.cpp:26) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'layer2_out.V' (firmware/myproject.cpp:62) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'layer3_out.V' (firmware/myproject.cpp:66) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'layer4_out.V' (firmware/myproject.cpp:70) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'layer5_out.V' (firmware/myproject.cpp:74) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'layer6_out.V' (firmware/myproject.cpp:78) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'layer7_out.V' (firmware/myproject.cpp:82) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'layer8_out.V' (firmware/myproject.cpp:86) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'b2.V'  in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'mult.V' (firmware/nnet_utils/nnet_dense_latency.h:39) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'acc.V' (firmware/nnet_utils/nnet_dense_latency.h:40) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'exp_res.V' (firmware/nnet_utils/nnet_activation.h:245) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'left.V' (firmware/nnet_utils/nnet_common.h:52) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'right.V' (firmware/nnet_utils/nnet_common.h:53) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'left.V' (firmware/nnet_utils/nnet_common.h:52) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'right.V' (firmware/nnet_utils/nnet_common.h:53) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'b8.V'  in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'mult.V' (firmware/nnet_utils/nnet_dense_latency.h:39) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'acc.V' (firmware/nnet_utils/nnet_dense_latency.h:40) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'b6.V'  in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'mult.V' (firmware/nnet_utils/nnet_dense_latency.h:39) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'acc.V' (firmware/nnet_utils/nnet_dense_latency.h:40) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'b4.V'  in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'mult.V' (firmware/nnet_utils/nnet_dense_latency.h:39) in dimension 1 completely.
INFO: [XFORM 203-101] Partitioning array 'acc.V' (firmware/nnet_utils/nnet_dense_latency.h:40) in dimension 1 completely.
/tools/Xilinx/Vivado/2020.1/bin/loader: line 286:  2103 Killed                  "$RDI_PROG" "$@"

I am lost as to why this is occurring.
Can anyone help me fix the problem?

thesps, 2021-09-06:
Hi, it looks like your Vivado HLS synthesis process was killed, probably because it ran out of memory.

> /tools/Xilinx/Vivado/2020.1/bin/loader: line 286: 2103 Killed "$RDI_PROG" "$@"

bluepra, 2021-09-06:
How do you suggest I fix this? I am running all my scripts on an AWS EC2 t2.medium instance with 4 GB memory.

thesps, 2021-09-07:
Since you're using cloud resources, sizing up to a larger instance would be a good place to state. I think a t2.large instance will be enough.

bluepra, 2021-09-09:
Switching to t2.large fixed the issue. Thank you!

closed 2021-09-09 by bluepra