fastmachinelearning/hls4ml #393: ZeroPadding2D Layer

LordScarface, 2021-09-05:
Hi, I'm trying to convert a Model containing a ZeroPadding2D Layer. From the Code it looks like that layer should be supported, however when generating the config for the Model, the following Exception occures: 

`ERROR: Unsupported layer type: ZeroPadding2D` 

I'm on version 0.5.1 of hls4ml

Regards,
Lukas

jmduarte, 2021-11-08:
Could you try with the master branch?

LordScarface, 2021-11-09:
Thank you for the reply,

I still get the error when using the master branch and also on the new `release-v0.6.0` branch. The same goes for `UpSampling2D` and `DepthwiseConv2D` Layers, even though they are all listed here:

```python
test = hls4ml.converters.get_supported_keras_layers()
print('DepthwiseConv2D' in test)
print('UpSampling2D' in test)
print('ZeroPadding2D' in test)
print(test)
```
```
True
True
True
['GarNet', 'GarNetStack', 'QDense', 'QConv1D', 'QConv2D', 'QActivation', 'QBatchNormalization', 'QConv2DBatchnorm', 
'Flatten', 'Reshape', 'UpSampling2D', 'Permute', 'MaxPooling1D', 'MaxPooling2D', 'AveragePooling1D', 'AveragePooling2D', 
'GlobalMaxPooling1D', 'GlobalMaxPooling2D', 'GlobalAveragePooling1D', 'GlobalAveragePooling2D', 'Conv1D', 
'SeparableConv1D', 'Conv2D', 'SeparableConv2D', 'DepthwiseConv2D', 'Add', 'Subtract', 'Multiply', 'Average', 'Maximum', 
'Minimum', 'Concatenate', 'Dot', 'ZeroPadding1D', 'ZeroPadding2D', 'InputLayer', 'Dense', 'BinaryDense', 'TernaryDense', 
'Activation', 'LeakyReLU', 'ThresholdedReLU', 'ELU', 'PReLU', 'Softmax', 'ReLU', 'BatchNormalization']
```

berab, 2021-12-09:
I think you should just add ZeroPadding2D (and 1D) in case they are supported. @jmduarte

PR #480 (2022-01-12): support ZeroPadding layers
Small change to support ZeroPadding1D/2D layers 

Resolves #393
closed 2022-01-12 by thesps