fastmachinelearning/hls4ml #401: RTL to FPGA bitstream

bluepra, 2021-09-24:
Hello,
I used the hls_model.build() command to turn a 3 linear layer Keras model to its RTL version (I followed the tutorial). From my understanding, I now have the HDL version of the model. I think I need to use vivado to convert this HDL version to a bitstream that can be flashed onto an FPGA. Am I correct in saying this?

Two things:
1. Where is the HDL version of my model stored?
2. I could use some help to figure out how to use vivado to get the bitstream.

Please feel free to correct my understanding at any stage of the process.

Thank you!

thesps, 2021-09-27:
Hi,
For the tutorial projects you can find the generated RTL under, for example `model_1/hls4ml_prj/myproject_prj/solution1/impl/{verilog, vhdl}`. This is the RTL of the NN IP. To generate a bitstream that you can load onto an FPGA, you need to integrated that IP into a larger project - to connect it to the outside world somehow. You can check out the new `VivadoAccelerator` backend (currently on `master` branch, not yet in a release) to make bitstreams for Pynq-Z2 or ZCU102 boards. If you have a similar type of board like another Zynq SoC kit it should not be too hard to adapt.

vloncar, 2023-11-21:
- Don't use `hls4ml` from the command line, it is deprecated and doesn't support the full feature set. Use the python API. Theck the tutorial to see how.
- Don't post unrelated questions in old issues

closed 2023-11-21 by vloncar