fastmachinelearning/hls4ml #412: LSTM model for HLS generation

samifry, 2021-10-09:
Hi!

I've tried the demo of HLS generation for the LSTM model example. It works, but I need to know if the solution is supported for RTL generation and for FPGA implementation. I need to do that with a current LSTM network with a large set of data. Maybe there is other issues to do that ?

Thanks in advance!
Sami

TitechTraj, 2022-02-03:
Hi Sami,
I am trying to generate LSTM (and later GRU) and used the recursive2 branch too. Did you manage to synthesize the model with Vivado HLS?
The first step of converting the model ( python keras-to-hls.py -c keras-config.yml ) seems to work, but when I run vivado_hls, I end up on some errors:
`hls4ml_recursive2/nnet_utils/nnet_recursive.h:434:93: error: use of overloaded operator '<<' is ambiguous (with operand types 'ostream' (aka 'basic_ostream<char>') and 'ap_fixed<18, 8, 5, 3, 0>')`
and
`hls4ml_recursive2/nnet_utils/nnet_recursive.h:435:93: error: use of overloaded operator '<<' is ambiguous (with operand types 'ostream' (aka 'basic_ostream<char>') and 'ap_fixed<18, 8, 5, 3, 0>')`

I am using Vivado 2019.2 and since this branch is old, I am wondering if it is a version issue. Did you run into a similar problem?

Thanks for sharing

TitechTraj, 2022-02-03:
By the way, I tried to run the code from the other repository ( https://github.com/richarao/hls4ml/tree/keras-lstm) and used the following python script:
```python
import yaml
import hls4ml

#load yml file
with open('keras-config.yml') as f:
    yamlConfig = yaml.load(f, Loader=yaml.FullLoader)

if __name__ == "__main__":
    model=hls4ml.converters.keras_to_hls(yamlConfig)
```
It worked for me and had the output:
```
Interpreting Model
Topology:
Layer name: input_1, layer type: InputLayer, current shape: [[None, 20, 6]]
Layer name: lstm_lstm, layer type: LSTM, current shape: [[None, 20, 6]]
Layer name: rnn_densef, layer type: Dense, current shape: [[None, 120]]
Creating HLS model
```
I am wondering if your problem is with using a different hls4ml branch that does not have LSTM defined. Did you try to install python setup.py install from the root of https://github.com/richarao/hls4ml/tree/keras-lstm ?

TitechTraj, 2022-02-03:
Update on my HLS synthesis issue. It was just an incompatible version. Both projects were generated using Vivado HLS 2018.2 successfully.

samifry, 2022-02-03:
Hi @TitechTraj ,

Thank you for your return.

I also fixed my problem, I had the official version of hls4ml installed and this provided some conflicts. If ever, I've used this branch https://github.com/chaitanyaPaikara/hls4ml/tree/keras-RNN for LSTM conversion and I can confirm that the results are satisfactory in the final hardware level. This is just a more updated version than this https://github.com/richarao/hls4ml/tree/keras-lstm

TitechTraj, 2022-02-03:
Oh great, thanks for sharing. I haven't implemented the hardware on the board but good to know that the new branch has more updated code. I will try it tomorrow.

closed 2022-02-03 by samifry

ntsakoulis, 2022-06-18:
Is it possible to generate HLS project with GRU or LSTM layers

TrajM, 2022-06-18:
If your network is not too large, I think it works. For my design I have 2 GRU cells with 120 hidden units and an input larger than 512, it didn't fit and I gave up, you would need a lot of manual HLS optimizations (for loop unrolling and array partitions).
I updated the RNN branch with hls4ml v0.6.0 to be able to use the pynq accelerator template (as explained in the official tutorials)
Give it a try and see if it works for you:

https://github.com/TitechTraj/hls4ml/tree/keras-RNN

TitechTraj, 2022-06-18:
If your network is not too large, I think it works. For my design I have 2 GRU cells with 120 hidden units and an input larger than 512, it didn't fit and I gave up, you would need a lot of manual HLS optimizations (for loop unrolling and array partitions).
I updated the RNN branch with hls4ml v0.6.0 to be able to use the pynq accelerator template (as explained in the official tutorials)
Give it a try and see if it works for you:

https://github.com/TitechTraj/hls4ml/tree/keras-RNN