fastmachinelearning/hls4ml #417: Vivado HLS generating multiple output streams

LordScarface, 2021-10-15:
Hello again,

when synthesizing my design, Vivado HLS breaks up the result vector into individual AXI streams. I would assume that the intended behavior would be to have one input and one output stream to the IP Core.

In my mnist example, HLS generates 10 individual AXI streams for the return Values:
[image]

I've also seen my other projects behave in the same way. Is this an issue with the Vivado HLS Version I'm using (2020.1) or do I have something configured incorrectly in Vivado HLS?

I would appreciate some help with this,
Best Regards,
Lukas

closed 2022-04-25 by LordScarface

bo3z, 2026-02-09:
Can you try out the Vitis Accelerator (#991) or Coyote Accelerator backends (#1347). They work with Vitis backends.

bo3z, 2026-02-10:
They are not merged yet. You will have to check out the branch from the respective PRs mentioned above and install hls4ml locally (by doing cd hls4ml && pip install .)

bhitesh311, 2026-02-11:
I did that. But I guess only alveo-u55c is supported. I happen to use PYNQ-Z2.