fastmachinelearning/hls4ml #418: VHDL Input understanding

Lsabetta, 2021-10-19:
Good morning,
[QS916.py.zip](https://github.com/fastmachinelearning/hls4ml/files/7371825/QS916.py.zip)

I managed to go through implementation of a Conv2D model till the end, and also to export the firmware as an IP..

The issue now is I’m trying to understand the VHDL output produced by vivado_hls. I managed to import the project into vivado, but still can’t figure out what is what since I'm not able to do a matching between what I see as signal names and what I would expect. I need to preprocess some data in VHDL and understand how to pass them the NN. Do you have any insight on how to interpret each signal in vivado HLS? Also, do you know if a test bench is also produced?

I've attached the python file which produces the design.

Thanks in advance

thesps, 2022-01-26:
Could you share the port descriptions you have?

Depending on the design you're integrating into, you may also try the `VivadoAccelerator` backend to target some supported boards. That will wrap the NN in another layer presenting a standard AXI-Stream interface.

In your line `convert_from_keras_model` you can add `backend='VivadoAccelerator'` and `board='zcu102`' (for example) . If your preprocessing can be packaged as an IP with AXI interfaces, it should then be possible to interface to the NN with a block design with Vivado IPI (as we do in the scripts in that backend).

Lsabetta, 2022-01-28:
Dear Sioni,
This is the port description:

entity myproject is
port (
    input_student_V_data_0_V_TDATA : IN STD_LOGIC_VECTOR (15 downto 0);
    layer20_out_V_data_0_V_TDATA : OUT STD_LOGIC_VECTOR (15 downto 0);
    layer20_out_V_data_1_V_TDATA : OUT STD_LOGIC_VECTOR (15 downto 0);
    const_size_in_1 : OUT STD_LOGIC_VECTOR (15 downto 0);
    const_size_out_1 : OUT STD_LOGIC_VECTOR (15 downto 0);
    ap_clk : IN STD_LOGIC;
    ap_rst_n : IN STD_LOGIC;
    const_size_in_1_ap_vld : OUT STD_LOGIC;
    const_size_out_1_ap_vld : OUT STD_LOGIC;
    ap_start : IN STD_LOGIC;
    ap_done : OUT STD_LOGIC;
    input_student_V_data_0_V_TVALID : IN STD_LOGIC;
    input_student_V_data_0_V_TREADY : OUT STD_LOGIC;
    layer20_out_V_data_0_V_TVALID : OUT STD_LOGIC;
    layer20_out_V_data_0_V_TREADY : IN STD_LOGIC;
    layer20_out_V_data_1_V_TVALID : OUT STD_LOGIC;
    layer20_out_V_data_1_V_TREADY : IN STD_LOGIC;
    ap_ready : OUT STD_LOGIC;
    ap_idle : OUT STD_LOGIC );
end;

I don't know exactly what each signal represents and how I am  expected to pilot them in the tb.

I tried to do as you suggested and the implementation with HLS4ML/Vivado_hls runs successfully. Problem is I don't really know what/how to do it next. Would it be possible for you to point me to a guide/example of a working test bench for an HLS4ML project?

Thanks a lot in advance,
Luigi

thesps, 2022-01-31:
The [UG902](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2020_2/ug902-vivado-high-level-synthesis.pdf) section "Interface Synthesis" is the best resource for this. There is a diagram on page 80 with the required timing of the control signals for this default interface. But it's worth noting that you can control the interface with pragmas in the HLS, so you may want to modify things depending on your application.

Regarding the testbench, we only created the HLS testbench [here](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/templates/vivado/myproject_test.cpp), but you could take a look at the HDL testbench that Vivado HLS generates from this after running cosimulation.

closed 2022-06-19 by jmduarte

gioregg, 2024-09-02:
Hi @Lsabetta ,

I am in your same situation. Did you find a solution for interpreting the signals in Vivado and for creating a testbench?

Thank you,

Gio