fastmachinelearning/hls4ml #431: AttributeError: module 'hls4ml.converters' has no attribute 'convert_from_onnx_model'

AlexMontgomerie, 2021-11-01:
I am having issues loading a model from onnx to hls4ml. I've tried running the following:

```python
import onnx
import hls4ml

model = onnx.load("example.onnx")
hls_model = hls4ml.converters.convert_from_onnx_model(model)
```

However I get the following error message:

```
AttributeError: module 'hls4ml.converters' has no attribute 'convert_from_onnx_model' 
```

It seems to be implemented in the repo, however it doesn't seem to be imported: https://github.com/fastmachinelearning/hls4ml/blob/bdd0bc5a589d9480ab1f443444531d32d525a082/hls4ml/converters/__init__.py#L301-L303

The version I am using is `0.5.0`. Please let me know if there's a solution for this!

ZhangYuQAQ, 2021-11-02:
Hello, I also encountered similar problems. When I ran the provided pytorch model, I also encountered the following problems: **attributeerror: module 'hls4ml. Converters' has no attribute' pytorch_ to_ hls'**

thesps, 2021-11-02:
On `0.5.0` there isn't a `convert_from_onnx_model`, but there is `convert_from_config` that is able access the `onnx` converter. The `onnx` converter has had an overhaul on the master branch which will be in the next release.

You could install from the master branch like `pip install git+https://github.com/fastmachinelearning/hls4ml.git`, or clone the repository and install with `pip install .`.

You'll also need to make sure that `onnx` is installed in the same Python environment as `hls4ml`.

AlexMontgomerie, 2021-11-03:
Thanks for your reply, I'm having some issues with this approach, I've created a simple configuration dictionary based on an example in the `converters/__init__.py` file, like so:

```python
    # create a configuration file                                                                                                                                                                         
    config = {                                                                                                                                                                                            
        "OnnxModel" : "example.onnx",                                                                                                                                                                    
        "OutputDir" : "outputs/hls4ml",                                                                                                                                                                        
        "ProjectName" : "hls4ml_test",                                                                                                                                                                  
        "XilinxPart" : "xcku115-flvb2104-2-i",                                                                                                                                                            
        "ClockPeriod" : 5,                                                                                                                                                                                    
        "IOType" : "io_stream",                                                                                                                                                                          
        "HLSConfig" : {                                                                                                                                                                                   
            "Precision" : "ap_fixed<16,6>",                                                                                                                                                             
            "ReuseFactor" : 1                                                                                                                                                                     
        }                                                                                                                                                                                             
    }                                                                                                                                                                                                 
    hls_model = hls4ml.converters.convert_from_config(config)  
```

however I'm getting the following error:

```
Exception: No precision for data->accum found and no default specified. 
````

> Also, it only appears to work for `onnx==1.8.1`

thesps, 2021-11-03:
I think it's almost right, but this section:

```
        "HLSConfig" : {                                                                                                                                                                                   
            "Precision" : "ap_fixed<16,6>",                                                                                                                                                             
            "ReuseFactor" : 1                                                                                                                                                                     
        }  
```

should instead be:
```
        "HLSConfig" : {         
            "Model" : {                                                                                                                                                                       
                "Precision" : "ap_fixed<16,6>",                                                                                                                                                             
                "ReuseFactor" : 1 
            }                                                                                                                                                                    
        }  
```

AlexMontgomerie, 2021-11-03:
thank you, it works now!

closed 2021-11-03 by AlexMontgomerie