fastmachinelearning/hls4ml #433: Two questions：1、Attributeerror: module 'hls4ml. Converters' has no attribute' pytorch_ to_ hls'

ZhangYuQAQ, 2021-11-02:


ZhangYuQAQ, 2021-11-02:
2、I directly use the following statement to install hls4ml: **pip install hls4ml**. The downloaded code is not the latest version 0.5

thesps, 2021-11-02:
> hls4ml: pip install hls4ml. The downloaded code is not the latest version 0.5

Can you double check the `hls4ml` version? `pip install hls4ml` should fetch the latest version from PyPI, which is `0.5.0`. There is no `pytorch_to_hls` in `hls4ml v0.5.0`, but you can `convert_from_config`.

We do have `convert_from_pytorch_model` in master branch, but not yet in a release. You can install that with `pip install git+https://github.com/fastmachinelearning/hls4ml.git` or git clone the repo and `pip install .` from there. Make sure that `torch` is installed in the same Python environment as `hls4ml`.

closed 2021-11-10 by ZhangYuQAQ