fastmachinelearning/hls4ml #436: Dense Layer with Linear activation causes clone stream later not working

ChiRuiChen, 2021-11-03:
So I tried a simple model with a Dense layer with BatchNorm afterward that has multiple outputs. 
I know the model doesn't make sense, I just want to try out the clone stream function.

[image]

[image]

So I print out the model's streams variables
[image]

Usually, it should create clone stream variables, but instead, the  2 BatchNorm layers after the dense layer are skipped.
Therefore comes out this error:
[image]

The problem is solved after I set the activation function of the Dense layer to RELU.

Any suggestion?

thesps, 2021-11-03:
Hi, first thing: which hls4ml version are you using? The latest release (v0.5.0) from PyPI, or the master branch from GitHub?

ChiRuiChen, 2021-11-04:
> Hi, first thing: which hls4ml version are you using? The latest release (v0.5.0) from PyPI, or the master branch from GitHub?

I use the latest master branch

thesps, 2021-11-04:
I didn't get a chance to look into this properly yet. I wondered if it may be due to the [`FuseConsecutiveBatchNorm`](https://github.com/fastmachinelearning/hls4ml/blob/master/hls4ml/model/optimizer/passes/qkeras.py#L199) optimizer pass.

> The problem is solved after I set the activation function of the Dense layer to RELU.

But this would suggest it's something else.

ChiRuiChen, 2021-11-06:
> I didn't get a chance to look into this properly yet. I wondered if it may be due to the [...]

Could you try out building the same model? Cause I think Dense Layers with multiple outputs are quiet common

jmduarte, 2021-12-26:
@ChiRuiChen, sorry for the delay.

This particular issue can be resolved by explicitly skipping the batchnorm fusion optimizers like this:

```python
inputs = Input(shape=in_size, name='input_1')
x = Dense(64)(inputs)
x = BatchNormalization()(x)
x = BatchNormalization()(x)
x1 = BatchNormalization()(x)
x2 = BatchNormalization()(x)
outputs = Concatenate()([x1, x2])
model = Model(inputs=inputs, outputs=outputs)

# Make an hls4ml config & model
config = hls4ml.utils.config_from_keras_model(model, granularity='name')
config['SkipOptimizers'] = ['fuse_batch_norm', 'fuse_consecutive_batch_normalization']

# Make Model
hls_model = hls4ml.converters.convert_from_keras_model(model,
                                                       hls_config=config,
                                                       io_type=io_type,
                                                       output_dir='hls4mlprj_model',
                                                       part='xcvu9p-flgb2104-2-i')
```

But, I also opened a PR, which will skip the optimizers when it's detected that the same inputs/outputs are used multiple times.

There's a further issue here that we need a `concatenate1d` function for `io_stream`, but that should be an easy feature to add.

PR #481 (2022-11-02): skip BatchNorm fusion when input/output is used multiple times
Skip BatchNorm fusion when input/output is used multiple times

Partially resolves #436
closed 2022-06-17 by vloncar